libopencm3/libopencm3 #704: Writing a "high-level" library based on libopencm3

Salamandar, 2016-10-02:
Hi there,
Reading the current libopencm3 examples, that contain a lot of shared code base, and seeing the current non-existence of a libopencm3-based high-level "user friendly" library, I'm starting to write one, partly based on the libopencm3 examples, partly based on https://github.com/MaJerle/stm32f429/ (Not very well written but that's still a good base).
The repo is here : https://github.com/Salamandar/stm32Libs/ (will probably be renamed).
The "TODO issue" is there : https://github.com/Salamandar/stm32Libs/issues/1
Please feel free to {add some to the todo, contribute, criticize}.

karlp, 2016-10-02:
This is the sort of thing that is probably better as links in the wiki.  When you have a little more to link to, I think it would totally reasonable to add on this page: http://libopencm3.org/wiki/Projects_based_on_libopencm3

closed 2016-10-02 by karlp

kuldeepdhaka, 2016-10-03:
@Salamandar Are you aware of https://github.com/insane-adding-machines/unicore-mx ?
`Writing a "high-level" library` is one of the aim.

Salamandar, 2016-10-03:
@kuldeepdhaka I dont see any high-level code in unicore-mx (at least not more high-level than opencm3) :) But thanks for the link, i'll check it out !