libopencm3/libopencm3 #707: STM32F4 clock setup operation order

Cat-Ion, 2016-10-18:
In `rcc_clock_setup_hse_3v3()` there's a [comment regarding the order of operations](https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/f4/rcc.c#L731) and a "TODO: why?".

I think it's because of the note in the reference manual about upper limits to the APBx clock speeds (45 and 90 MHz, respectively). This would mean that doing it like this might be incorrect, as calling the setup again with lower prescaler values could increase the clocks above those limits.

Example:
- Setup using `rcc_hse_8mhz_3v3[RCC_CLOCK_3V3_168MHZ]`. The main PLL will run at 168 MHz, APB1 and 2 at 42 and 84 MHz.
- Call setup again, with `rcc_hse_8mhz_3v3[RCC_CLOCK_3V3_84MHZ]`. The new prescaler values are 2 and 1, so between the calls to set these prescalers and the main PLL setup, the APB1 and 2 clocks will run at 84 and 168 MHz, which exceeds the limits given in the reference manual.

A solution might be to set hpre/ppre1/ppre2 to the higher of the old and new prescaler and then to the new one after the PLL has been touched.

ChuckM, 2016-10-18:
You are correct. There is a lot of improvement that could be done in the rcc.c code. @karlp did some updates to help keep things in sync during changes to the power profile, it could delete or update this comment.

Cat-Ion, 2016-10-18:
Do those updates make sure that the limits aren't exceeded? I haven't seen anything like that while looking around, but I've only spent a few hours with stm32s so far, so overlooking something wouldn't be a big suprise.

If they don't I've got some code that does that, and updates the comment as well. Just wanted to make sure that it's actually needed before I push it and make a pull request. :)

karlp, 2016-10-19:
The updates @ChuckM is talking about is  https://github.com/libopencm3/libopencm3/commit/095ed8511a82e7f7da5c59c42ca116bb228a20d6 which was only applied to L1, as it was the only one I could test that one with reliably at the time.  But yes, _all_ the clock setup routines are "vulnerable" to this.

One way of approaching this is that those helper routines are only really intended for use for initial startup, going from reset up to full operating speed.  If you have more particular needs, such as dynamically changing clocks, you're almost guaranteed of needing to use the individual piece functions anyway, rather than just the helpers.  See for instance https://github.com/libopencm3/libopencm3/issues/685 which is functionality not even provided by the helpers.

karlp, 2016-10-19:
Further, adding code to these routines to handle all the dynamic cases makes the code bigger for alllll users

Cat-Ion, 2016-10-19:
Yeah, I didn't really have a /problem/ with it, I was just digging into a new (for me) platform and came across it. :)

The code I was thinking of isn't big, just two copies of `if(new_prescaler > old_prescaler) set_prescaler(new_prescaler);` for APB1 and APB2, but if you're saying that the helper function is just for startup that shouldn't be necessary. How about just adding a comment to the documentation?

karlp, 2016-10-19:
They're probably both suitable, they just need to be done, and ideally should be done consistently.  It's also the sort of thing that _really_ should be done by actual testing some perverse paths, not just reading the ref man and going, "I think it should be ok to do it in this way"  We really want to avoid code that is only there because people _think_ it might be necessary.

If you want to work on this, consider something like https://github.com/karlp/libopencm3-tests/blob/master/tests/rcc-legal-ranges/main-stm32l1-generic.c to verify first that there is actually a problem, and secondly that any changes made actually fix that problem :)

(That project contains tests that aren't cross-target enough (yet) to put into the main libopencm3/tests directory, or that I haven't worked out a way of automatically collecting the test results yet)