libopencm3/libopencm3 #724: usb:event (reset,sof,suspend,resume) events have no caller device parameter

alexrayne, 2016-12-09:
Hallow.
callbacks reset,sof,suspend,resume for usb_device are void(*)(void) type, they have no it`s own usb_device* parameter. therefore for multiple usb ports it must be provided multiple callbacks on event,  or other complexity. there is no proplem to pass one parameter from usbd core. (imho - such parameter even can not be conflicted with old parameterless handlers)
is there maintainers interested for change such API? i can provide an patch, it not difficult.

karlp, 2016-12-09:
is the motivation here to use the same callback function for both devices?  Adding a param probably doesn't help, but I'm not sure I buy the "reduced complexity" of then knowing that your single callback function can get called from two different usb devices.  You're very free to implement a combined callback in your application code if you feel that's easier to follow.  I'm not against this particularly, but I don't see it as any particular gain.

kuldeepdhaka, 2016-12-09:
Looks like this is what i have done in past. :)

https://github.com/insane-adding-machines/unicore-mx/blob/master/include/unicore-mx/usbd/usbd.h#L169-L177 (look in the commit history, it has many more cool+fixed stuff)

Also, The PR originally died in libopencm3: https://github.com/libopencm3/libopencm3/pull/485

alexrayne, 2016-12-09:
Thank you kuldeepdhaka, i`dont understand - what is the project your refer? - it is not a fork of opencm3? and what about your pull - is it accepted somwhere?

karlp:  yes, the motivation is one - share same code on all devices. without this, need to hardcoding for particular platform without scalability. it penaltyes by more code.
this parameter actualy presents in usbd core routines that invokes one callbacks, it need only pass it. without this parameter, user of a library enforced to write more code.

compatibility with old code, imho, may be provided by configuraniot marco - for example $DGENERIC_CALLBACK_STYLE=OLDSTYLE, for old projects by defult, and explicitly note NEWSTYLE for projects that need multiple devices support.

kuldeepdhaka, 2016-12-09:
@alexrayne The PR was merged in unicore-mx (fork of libopencm3) [Including numerous bug fixes].

alexrayne, 2016-12-09:
kuldeepdhaka: can i just substitute ocm3 library by unicore-mx or it have different API? how difficult migration?

kuldeepdhaka, 2016-12-10:
@alexrayne Functional drop in compatible (minor change in api to accomodate new feature).
You can see Example repo: https://github.com/insane-adding-machines/unicore-mx-examples
Migration will be easy, use the above examples repo to see working demo and make changes accordingly.

If you are super excited, you can use the new USBD stack (Redesigned altogether) instead. 
See PR: https://github.com/insane-adding-machines/unicore-mx/pull/26

alexrayne, 2016-12-10:
kuldeepdhaka: i saw at that project - they are dropoff origin usb stack - imho it is a miss, cause of incompatibility with projects based on original ocm3, and forces user to make rework for migration to unicores.
as for your usbd stack: 
1) i saw that you split from usbd_device backend and backend_data. why you place that structures out of usbd_device? access to them a bit heavy vs native structure field. more - static declaration of usbd_device now more complex - requres co-declaration of that backends structures.
Where this stack can be discussed?

kuldeepdhaka, 2016-12-10:
@alexrayne We can discuss on IRC Freenode #unicore-mx (http://webchat.freenode.net?channels=%23unicore-mx)

alexrayne, 2016-12-10:
kuldeepdhaka:  how to use it? i`ve login some posts. after relogin no any history no any conversation comes

kuldeepdhaka, 2016-12-10:
@alexrayne You can install a IRC client (example: hexchat). it will store history on your HDD + other features.