libopencm3/libopencm3 #730: license question?

mgiaco, 2016-12-22:
Hello,

Please can someone give me an explanation about the usage of this lib for commercial sw. I would like to use this lib on an proprietary hardware with some closed source code. So I am not sure if I can use this lib or not - the license stuff is really a pain if you are not a lawyer :-).

It clear that if I made some changes I will contribute them back - but the other SW must be closed. I ask before I start with development.   

cheeers

Respekt1, 2017-01-06:
I also would like to use this in a similar way - can someone give here any explanation.

many thanks

karlp, 2017-01-10:
The shortest, least helpful answer is, "talk to your lawyer"

The slightly more useful answer is that you need to make sure that anyone you distribute to, has the ability to relink against a modified libopencm3.  Remember, the LGPL is so that that  portion can be modified for bugfixes/enhancements by the user.  You can do this by such means as (non-exhaustive list) 

- providing all  your own code as open source
- providing .o/.a files for your private application and a suitable build script to recreate the necessary binary
- probably other methods

It's important to remember who and what counts as distributing too.  First and foremost, you must make this decision yourself, with your own needs and requirements in mind.

closed 2017-01-10 by karlp

ohhai, 2021-01-20:
It would be great to have such details mentioned in readme. Law compliance between LGPL and other (open source) licenses is a complicated question, so direct official guides from libopencm3 would be very helpful.
I came here because of similar question of using libopencm3 in my project under EUPL. FSF states they are incompatible (at least directly), this fact might scare some users.