libopencm3/libopencm3 #736: absent file <libopencm3/stm32/f1/nvic.h>

NikolayBobovnikov, 2017-01-12:
Hello,
I've just found out about this project, and decided to check it for STM32F103. This is the code:

```
#include <libopencm3/stm32/rcc.h>
/* Set STM32 to 72 MHz. */
static void clock_setup(void)
{
    rcc_clock_setup_in_hse_8mhz_out_72mhz();

    /* Enable GPIOB, GPIOC, and AFIO clocks. */
    rcc_periph_clock_enable(RCC_GPIOB);
    rcc_periph_clock_enable(RCC_GPIOC);
    rcc_periph_clock_enable(RCC_AFIO);
}
int main(void)
{
    clock_setup();
    while (1) {
    }
}
```
Compilation fails with the message: libopencm3/stm32/f1/nvic.h: No such file or directory
And there is actually no nvic.h file in libopencm3/stm32/f1 directory.

This is relevant part  in cm3/nvic.h:
```
#elif defined(STM32F1)
#	include <libopencm3/stm32/f1/nvic.h>
#elif defined(STM32F2)
```

karlp, 2017-01-12:
run make first. Some files are generated.

closed 2017-01-12 by karlp

karthikprabhu17, 2019-05-19:
I am facing the same issue..any workaround for this?