libopencm3/libopencm3 #738: Issue when building under Windows (cygwin)

NikolayBobovnikov, 2017-01-16:
Hello,

I'm facing an issue with building library on windows:
[image]

It seems that static library is not being created (no *.a file, which arm-none-eabi-ar gets as input)
I'm using Cygwin, with updated path variable:
[image]

Text of the error message:
```
D:\Software\arm_gcc\5_2_2015q4\bin\arm-none-eabi-ar.exe: /cygdrive/d/nbobovnikov/home/dev/auvir/auvir_embed/mocap/Libraries/libopencm         3/lib/libopencm3_stm32f0.a: No such file or directory
make[1]: *** [../../Makefile.include:36: /cygdrive/d/nbobovnikov/home/dev/auvir/auvir_embed/mocap/Libraries/libopencm3/lib/libopencm3         _stm32f0.a] Error 1
make: *** [Makefile:65: lib/stm32/f0] Error 2
```
Log file (although I doubt it would be useful):
[log.zip](https://github.com/libopencm3/libopencm3/files/708640/log.zip)

NikolayBobovnikov, 2017-01-16:
Looks like some path issue... I removed paths from the target (static lib) name as follows:
[image]

This helped. I'll need to figure out whats wrong with the paths, but right now I'll close the issue.

closed 2017-01-16 by NikolayBobovnikov

karlp, 2017-01-16:
You may try running with V=1 on the command line to see the full command lines used for each step, it might help debug where this went wrong.

ardyesp, 2017-01-22:
I had a similar issue with cygwin; had to manually build the targets in lib directory. Used following script:

#!/bin/bash

cd libopencm3

for i in lib/* ; do
  if [ -d "$i" ]; then
    echo "$i"
	for j in "$i"/* ; do
	  if [ -d "$j" ]; then
		echo "  $j"
		cd "$j"
		make clean || exit
		make || exit
		cd -
	  fi
	done
  fi
done

karlp, 2018-02-16:
Parts of windows builds at least was fixed via bf882caf4dde191c46fed89873fbbce2def587c6  @R030t1 could you just make a new issue with exactly what your problems are please?

dinicare, 2018-08-28:
Hi Guys,
I also try to build libopencm3 with Cygwin. It shows error just like you.
But when i read the README.md, it recommends to use msys shell to build it. And it works!