libopencm3/libopencm3 #739: Documentation for SPI on STM32F1 is not showing up

NikolayBobovnikov, 2017-01-19:
[Documentation](http://libopencm3.github.io/docs/latest/stm32f1/html/group__spi__file.html) on SPI module for STM32 F1 chip is not showing up when pressing "More..."

Since including "spi.h" in the project for F1 chip leads to including "spi_common_l1f124.h", I suppose [details page for F1](http://libopencm3.github.io/docs/latest/stm32f1/html/group__spi__file.html#details) should redirect to [page for L1](http://libopencm3.github.io/docs/latest/stm32l1/html/group__spi__file.html)?

Regards,
Nikolay

esden, 2017-01-19:
We are currently transferring the domains around. So the links you are referring to might not be working correctly for the next few days. I will try to look into it as soon as the dust settles a bit. :)

NikolayBobovnikov, 2017-01-19:
Ah, ok. At least this is known :)

karlp, 2017-07-04:
still broken, looks to just be more doxygen bustage

karlp, 2017-10-18:
This should all be fixed now, I've overhauled the doxygen generation to be based on the sources compiled.

closed 2017-10-18 by karlp