libopencm3/libopencm3 #740: SPI driver structure fro L0 L1 F0 F1 F2 F4

SevenW, 2017-01-20:
I am in the process of adding a STM32 L0 SPI driver. Once done, I like to make a pull request. I see a lot of similarity among the L's and F's, and now I am seeking advice on structuring the files, I compared "SPI register map" tables from various STM32 reference manuals. Compared to starting with spi_common_all.h the following types of STM32 add capabilities:

```
common_all < F1 < {L0, L1, F2, F4}
and:
common_all < F1 < F0 (this adds other functionality then the above)
```
Note: some low budget mcu's F100, F0x0 have fewer registers then common_all.

Observation on current implementation:
`common_all < L1F124 < F24`

What I want to do is separate out L1 and make a common L01F24

Please advice on the naming of the spi_common_*.h

fenugrec, 2017-01-20:
As I understand, I think the current direction is to drop the "L1F124" naming when it gets to a certain point of complexity. ADC and USB do this by grouping into arbitrary "versions", such as "st_usbfs_v1" and "st_usbfs_v2".

What are the registers "missing" from the low-end F1 and F0x0 chips you mention ?

(note, I'm mostly a spectator in this project, I don't make decisions ! )

SevenW, 2017-01-23:
I was in doubt on the _v2 naming convention. In tow cases, the _v2 suffix is an addition to the original _common_all (adc, usbfs) and in one case it is a full replacement (usart). 
Looking at this it seems appropriate to give the F1 version the name spi_common_v2.h

I will come back to the "reduced function" F1 and F0.

SevenW, 2017-01-23:
Missing from value line F0x0 and F100, with respect to other F0/F1:
```
register 0x1c: SPIx_I2SCFGR
register 0x1e: SPIx_I2SPR
flag b2: CHSIDE from register 0x08:SPIx_SR
flag b3: UDR from register 0x08:SPIx_SR
```

karlp, 2017-01-23:
it was my understanding that all L0 chips have the "spi v2+" peripheral, with the fifos, that comes in with f3 and f0.  There's probably not a lot you want to try and get too common.  Are you sure there's not a PR for spi on l0 already?

SevenW, 2017-01-23:
Is a PR a github "issue"? Could not find it.
I also could not find a pull request.

I have prepared and tested a L0 implementation. It is not a lot of change. I try to prepare a pull request tonight. (The only open question for me: should I rename spi_common_l1.h to spi_common_v2.h?)

SevenW, 2017-01-23:
A preview on my intended delta can be found here:
https://github.com/libopencm3/libopencm3/compare/master...SevenW:stm32spi

karlp, 2018-04-28:
tot he best of my knowledge this is closed now. new or extra or missing portions should come in as new tickets please.

closed 2018-04-28 by karlp