libopencm3/libopencm3 #741: Update generation of linker script to support C++

NikolayBobovnikov, 2017-01-21:
If project is using c++, there is a linking error which says 
```
`.eh_frame' referenced in section `.text' of /usr/lib/gcc/arm-none-eabi/4.9.3/armv7-m/crtbegin.o: defined in discarded section `.eh_frame' of /usr/lib/gcc/arm-none-eabi/4.9.3/armv7-m/crtbegin.o
```
It appears that in order to fix this, I need to remove this section from the generated linker script:
```
/*
* The .eh_frame section appears to be used for C++ exception handling.
* You may need to fix this if you're using C++.
*/
/DISCARD/ : { *(.eh_frame) }
```

I would be good not to have to alter generated linker script. For example, to have an option when building the library which says that I'm going to use C++, and this section is removed automatically.

This is not critical though, just a thought  :)

karlp, 2017-01-23:
Are you using exceptions with c++?  That tends to be disabled when building for embedded.  You would have seen a fairly noticeable size increase after keeping the .eh_frame data right?

It would definitely need to be included if you want exceptions, but I'm not really sure if supporting exceptions is something the core library code should be attempting to support.

NikolayBobovnikov, 2017-01-25:
No, I'm not using exceptions in the code, and I use options -fno-exceptions -fno-rtti for compilation, and I had had to remove the "discard" section in order to fix this linking error.  :)

I'm not even sure if decision to leave as is / remove of this discard section is possible during making the linker script (it is not known in advance if user will use exceptions)

>It would definitely need to be included if you want exceptions, but I'm not really sure if supporting exceptions is something the core library code should be attempting to support.

Agree. But if not-using-exceptions is by default, then is it better to remove this section by default, leaving comment like "if you want to use exceptions you should uncomment this /DISCARD/ section below"?

karlp, 2017-01-25:
Can you share your example project with it's makefiles please?  What toolchain are you using, and where did it come from?

NikolayBobovnikov, 2017-02-07:
Yes, sure. I'm using cmake with arm none eabi gcc/g++ compiler taken from [here](https://launchpad.net/gcc-arm-embedded/+download).

Compiler/linker flags are as follows:
```
SET(CMAKE_C_FLAGS   "-mcpu=cortex-m3 -mthumb -mabi=aapcs -Wall -std=gnu11 -fno-builtin -ffunction-sections -fdata-sections -fomit-frame-pointer -fno-unroll-loops -ffast-math -ftree-vectorize" CACHE INTERNAL "c compiler flags")
SET(CMAKE_CXX_FLAGS "-mcpu=cortex-m3 -mthumb -mabi=aapcs -Wall -std=c++14              -ffunction-sections -fdata-sections -fomit-frame-pointer -fno-unroll-loops -ffast-math -ftree-vectorize -fno-exceptions -fno-rtti -Wconversion" CACHE INTERNAL "cxx compiler flags")
SET(CMAKE_ASM_FLAGS "-mcpu=cortex-m3 -mthumb -x assembler-with-cpp" CACHE INTERNAL "asm compiler flags")
SET(CMAKE_EXE_LINKER_FLAGS "-Wl,--gc-sections -Wl,-Map=tut.map --specs=nosys.specs -fno-exceptions -fno-rtti -mthumb -mcpu=cortex-m3 -mabi=aapcs" CACHE INTERNAL "executable linker flags")
SET(CMAKE_MODULE_LINKER_FLAGS "-mthumb -mcpu=cortex-m3 -mabi=aapcs" CACHE INTERNAL "module linker flags")
SET(CMAKE_SHARED_LINKER_FLAGS "-mthumb -mcpu=cortex-m3 -mabi=aapcs" CACHE INTERNAL "shared linker flags")
```

Sorry for delay, I don't have much free time for my project currently. I'll make a mini project example on github reproducing this behavior and share the link here once ready.

karlp, 2017-02-07:
A couple of general comments.  you shouldn't need to provide -mabi=aapcs, you need to provide the --specs=nosys.specs to both the compielr and the linker, as per https://launchpadlibrarian.net/287100883/readme.txt

zyp, 2017-02-07:
Some comments from me as well. karlp is right in that the flags aren't the recommended ones, but I'm not sure those particular flags makes a difference. The path `/usr/lib/gcc/arm-none-eabi/4.9.3/armv7-m/crtbegin.o` already suggests that the right multilib target (armv7-m) is being picked.

What I'm wondering is whether crtbegin.o should even be linked at all. Since the .eh_frame section is referenced by the .text section in the same file, I'd say that this error would always happen when that file is combined with the libopencm3 linker script, regardless of whether it's being linked with C++ or just plain C code. My guess is that your linker flags are wrong, causing the linker to pull in a file that you don't want. I suggest you find a working example and compare its linker command to your linker command and sort out the differences.

Aside from that, GCC 4.9.3 is fairly old, so you'd probably want to upgrade that. Especially when working with C++14.

karlp, 2017-02-08:
crtbegin should have been avoided by nosys I would have thought, but that does need to be provided to both compiler and linker, and needs to be passed all the time.  I don't know what cmake is doing with which of your flags.

adamheinrich, 2017-11-29:
Just add `-nostartfiles` to linker options (probably `CMAKE_EXE_LINKER_FLAGS` in your case).

karlp, 2017-11-29:
Can we close this then? All the examples are built with -nostartfiles, so if you're not using that, that's "user error" as far as "c++ support" goes.  Unless there's something _else_ for c++?

closed 2017-11-29 by karlp

ChuckM, 2017-11-29:
Or is it that we could insure that the linker files exported the necessary variables so that when people were not using the `-nostartfiles` option their code still worked as expected?

stefanct, 2022-10-30:
FWIW, for me this also happens with plain C w/o `-nostartfiles` when supplying `-specs=nano.specs -specs=nosys.specs` in `CFLAGS` and `LDFLAGS`. If I am not mistaken this is not what karl expected?

karlp, 2022-10-31:
if people are not using -nostartfiles, then they're going to have a whole bunch of other problems anyway, like, "whose start files are running"  As libopencm3 does provide reset handlers and vectors.  I'm not really convinced that doing _anything_ to try and support that environment is productive for anyone.    Re-open this with some arguments if people think differently.

@stefanct yes, that's what I expect,  -nostartfiles is effectively required for doing a bare metal project as libopencm3 is built for.

stefanct, 2022-10-31:
I was/am just confused because I interpreted https://github.com/libopencm3/libopencm3/issues/741#issuecomment-278295351 to mean that `nosys.specs` automatically sets `-nostartfiles` and https://github.com/libopencm3/libopencm3/issues/741#issuecomment-347976803 to mean `-nostartfiles` is only required for c++.

I haven't looked if anything is already there but maybe a summary of this thread/your last message could be added somewhere to make it clear that `-nostartfiles` is indeed always required.

karlp, 2022-10-31:
nostartfiles = no startup files.  https://gcc.gnu.org/onlinedocs/gcc/Link-Options.html  (scroll down a bit, can't link directly)  that's required if you want to use libopencm3's startup and reset handler and vector handling. You don't _have_ to, but we're not going to try and provide support for that natively.  (You _can_ link in the "library" code and use it in a normal cmsis project, but it's confusing for just about everyone, and I really don't see any motivation for trying to support it)

--specs=nosys.specs is...
```
karlp@beros:~$ cat tools/gcc-arm-none-eabi-8-2018-q4-major/arm-none-eabi/lib/nosys.specs
%rename link_gcc_c_sequence                nosys_link_gcc_c_sequence

*nosys_libgloss:
-lnosys

*nosys_libc:
%{!specs=nano.specs:-lc} %{specs=nano.specs:-lc_nano}

*link_gcc_c_sequence:
%(nosys_link_gcc_c_sequence) --start-group %G %(nosys_libc) %(nosys_libgloss) --end-group

```
so, as covered https://stackoverflow.com/questions/65453293/what-are-nosys-nano-rdimon-terms-when-using-arm-gcc as well, "nosys" is about the _rest_ of the c runtime.