libopencm3/libopencm3 #746: EXTI object not included in STM32L0 (and STM32L4) Makefile

SevenW, 2017-02-15:
Although the header file support for nvic and exti is present for the STM32L0, the source file is not compiled and the object is not present.

Is this as easy as adding exti_commom_all.o the the L0 Makefile? 

I observed that it also is missing in the L4 makefile, but it is present for L1 and F0, F1, F2, F3, F4.

karlp, 2017-02-15:
probably, but it hasn't been added because it hasn't been tested.  The header support is done by copying out of the refman.  using exti_common_all requires comparing whether the exti implementaiton is the same.  (It probably is) and then adding the objects to the build.  Add it locally, see if it works, and we'll merge it :)

karlp, 2017-02-17:
merged in 8a62618

closed 2017-02-17 by karlp