libopencm3/libopencm3 #754: genlink script broken for STM32L4x6

ChuckM, 2017-03-08:
These parts split their 128K of RAM into a 96K part at 0x20000000 and a 32K part at 0x10000000, the current defined just says RAM=128K which generates a linker script assuming it is contiguous from 0x20000000. 

Quick fix is to change the RAM value to 96K. 

With it broken like it is, if you use it to generate an linker script your program will immediately bus error when it enters main() because it trys to push to stack a return address into non-memory.

karlp, 2017-03-08:
Can you check this fix please?
https://paste.fedoraproject.org/paste/klNw0~tEVGEQrlGDWuFiL15M1UNdIGYhyRLivL9gydE=

https://paste.fedoraproject.org/paste/5e6ROoTRBBHNb-0SiDbNDl5M1UNdIGYhyRLivL9gydE=/ is an example of the generated output.

I'm kinda curious though, https://github.com/karlp/libopencm3-tests/blob/master/tests/adc-power/ works on the l4 disco board I have, with the original devices.data.

And it runs, l4 was one of the test targets.  *shrugs*  it's definitely more correct from the datasheet to have those two sections.  let me know and I'll get it merged in.

ChuckM, 2017-03-08:
👍 looks good to me.  I've tested it in my repo and it generates both RAM regions and the correct flash region.

ChuckM, 2017-03-08:
Interesting that it worked for you.

closed 2017-03-30 by karlp