libopencm3/libopencm3 #755: USB endpoint callback only called every other bulk transfer?

joh, 2017-03-08:
Hi, I've made a simple USB blink example which sets up a vendor specific device with a single bulk endpoint. The endpoint callback is set to toggle an LED on the STM32F4 Discovery board.

On the host side I have a simple python client based on libusb which opens the USB device, writes a single byte to the endpoint and exiting. I expect the LED to toggle every time I run this client.

What I'm seeing, however, is that I need to run the client *twice* in order to toggle the LED. It seems like the endpoint callback is only called *every other* bulk transfer. Is this expected, or am I missing something?

The code: https://github.com/joh/libopencm3-examples/commit/31f0bc6273dc9114b1e2124864d3cb00c767307c

joh, 2017-03-08:
Forgot to mention that if I perform several bulk transfers after the other inside a loop within the client, they all seem to make it to the endpoint callback.

karlp, 2017-03-16:
just for starters, it seems unexpected.  But I've not looked at this any further yet.

joh, 2017-03-16:
As suggested by @zyp on #libopencm3, I commented out the dev.set_configuration() line in usb_blink.py, and indeed the problem goes away.

zyp, 2017-03-16:
For reference, from IRC today:

```
< zyp> hmm, it's probably a data toggle issue
< joh> Yeah
< joh> data toggle?
< zyp> can you try one thing for me?
< zyp> comment out dev.set_configuration() and test again
< joh> Sure, will test when I get home :)
< zyp> highlight me when you've done so
< joh> Ok, thanks
< zyp> anyway, the data toggle mechanism is how usb keeps host and device in sync in case of lost data and retransmits
< zyp> there's two kind of data packet tokens; DATA0 and DATA1, and they alternate
< zyp> if a data packet is lost, the device wouldn't reply with an ACK to it, so the host will send it again
< zyp> but the host is not able to distinguish between that and if the ACK packet got lost after the data packet got delivered
< zyp> so if the ACK gets lost, the host will resend the same datapacket, the device will see that it received the same data toggle again, assume it's a retransmit and ACK but silently discard it, since it's already received
< zyp> if the data toggle stuff isn't handled right, you might end up in a situation where you only receive DATA0 or only DATA1 packets
< zyp> in other words every other packet
< zyp> there's some rules around resetting the data toggle state machine when endpoints are reset, i.e. on set_configuration() and similar, and there's probably a bug there somewhere
< zyp> which is why you see the issue only when your script is restarted, not when it runs a loop
…
< joh> zyp: So I tried commenting out dev.set_configuration(), and indeed the problem goes away
< zyp> good, data toggle issue for sure then
```

I'm not sure exactly where the bug is located though. I'm guessing that maybe the device resets the state machine due to the set_configuration(), but the host doesn't, and this might be an intentional feature on the host to cope with devices that doesn't handle this per the spec.

In that case, doing a more explicit reset on the host (clear stall IIRC) might be the proper solution, but simply not doing set_configuration() at the start of the script might also be acceptable.

karlp, 2017-03-16:
zyp's analysis looks good.  Possibly the device could check if the config set is the same as the current config and ignore it, but yeah, somethings's fishy.  perhaps just ignoring repeated set_config's in the usb stack?

kanflo, 2017-03-22:
Could it be that libopencm3 does not reset the data toggle to DATA0 in all the situations expected according to the standard? [Some googling](http://www.microchip.com/forums/m473721.aspx) tells me it should be done for set configuration, set interface and clear feature but I have not checked the code nor verified with the USB spec. Currently grep:ing for the data toggle in lib/usb :)

zyp, 2017-03-22:
Well, it's just that if the host always sends DATA0 and the device keeps expecting DATA1, I would expect all packets to get dropped, not every other.

However, if device always expects DATA0 and host alternates sending DATA0 and DATA1, only the DATA1 packets will get dropped.

I haven't checked the code myself either, but this seems like the most likely explanation.

karlp, 2018-08-20:
It's done in ep_setup for both st_usbfs and dwc_otg.  I don't know off teh top of my head whether clear feature will do that, but set config and set interface definitely will.  I've no idea what the host is doing.