libopencm3/libopencm3 #760: Fix DFU serial number generation

alyoshin, 2017-03-16:
Patch:
[0001-stm32-Fix-DFU-serial-number-generation.patch.gz](https://github.com/libopencm3/libopencm3/files/847111/0001-stm32-Fix-DFU-serial-number-generation.patch.gz)

Tested on STM32F2, STM32F3, STM32F4.

STM32F2, STM32F4: 8 hex digits of (UID0 + UID2)
	followed by 4 hex MSB digits of (UID1)
UID 012: 002d0030 3334510b 32313533
SerialNumber: 325e35633334

STM32F3: 8 hex digits of (UID0 + UID2)
	followed by 4 hex MSB digits of (UID2)
UID 012: 003b0041 50335706 20333333

Disassembled fragments for F2, F3, F4 with comments on USB serial number generation in ST DFU mode:
[f2-f3-f4_sysmem-uids-asm.tar.gz](https://github.com/libopencm3/libopencm3/files/847122/f2-f3-f4_sysmem-uids-asm.tar.gz)

karlp, 2017-03-16:
Need to get info from the submitter of this _as_dfu patch to try and work out why you both have such different implementations!

alyoshin, 2017-03-17:
On Fri, Mar 17, 2017 at 1:06 AM, Karl Palsson <notifications@github.com> wrote:
>
> Need to get info from the submitter of this _as_dfu patch to try and work out why you both have such different implementations!

I have noticed the difference in ST DFU USB serial number and number
returned by this function. Then I check F2, F3 and F4 disassembled
system memory (relevant parts of it for different MCU series was
attached in archive) and wrote this patch

F1 system memory code does not seems to use UID register at all.