libopencm3/libopencm3 #761: Data corruption in USB control transfer

joh, 2017-03-17:
I've set up a control request which reads data from the host. It seems that quite often, that data is corrupted.

Control request 0x13 reads 6 bytes from host, stores them in a buffer.
Control request 0x93 writes 6 bytes from buffer to host.

Test firmware + libusb client here: https://github.com/joh/libopencm3-examples/tree/5aadcddf0836360959dd28346b9b11ae3ab3dc82/examples/stm32/f4/stm32f4-discovery/usb_blink

Expected output from test program:
```
$ python usb_blink.py ctrl_data
ctrl_data
wrote 6 bytes: array('B', [1, 2, 3, 4, 5, 6])
read 6 bytes: array('B', [1, 2, 3, 4, 5, 6])
```

However, sometimes data is corrupted:
```
$ python usb_blink.py ctrl_data
ctrl_data
wrote 6 bytes: array('B', [1, 2, 3, 4, 5, 6])
read 6 bytes: array('B', [5, 6, 69, 34, 0, 0])
```

Run in a while loop to reproduce quickly. Tested on STM32F4 Discovery. 
I've verified in the debugger that the data is corrupted as received by the device.
[Control transfers apparently use "best effort delivery"](http://www.beyondlogic.org/usbnutshell/usb4.shtml#Control), but this can't be expected?

FWIW I've implemented the same read/write buffer functionality with bulk transfers, and I don't see any data corruption there...

zyp, 2017-03-18:
This is a duplicate of #600 and should be fixed in #672. You can test your code again with the updated code from #672 to verify.

This is probably another reminder that we should get #672 merged soon.

joh, 2017-03-18:
Just tested #672 and indeed it appears to fix the issue :)

closed 2017-03-19 by karlp