libopencm3/libopencm3 #763: GitHub repo description could use an edit

DavidEGrayson, 2017-03-27:
Hey, I just found out about this library today, and just wanted to quickly point out that the description for the GitHub repository, which GitHub shows at the top of the [main page](https://github.com/libopencm3/libopencm3), uses questionable capitalization and punctuation.

The current description is:

> Open Source ARM cortex m microcontroller library

I think we should write the names of ARM products the same [way](http://www.arm.com/products/processors/cortex-m) ARM does.  I would probably make "source" be lower-case too, or at least make it be the same case as "library".  So it should be:

> Open source ARM Cortex-M microcontroller library

This is not something I can make a pull request for.  It will help make the repository look more professional.

karlp, 2017-03-27:
Minor, but reasonable, and easy enough.  @esden I can't do this, I think you're the only one with perms for that.

karlp, 2018-04-28:
fixed. thanks for the report.

closed 2018-04-28 by karlp