libopencm3/libopencm3 #773: make does not build library with default makefile

stanislavvv, 2017-04-17:
commit 383fafc862c0d47f30965f00409d03a328049278 (current master)
make give me:

```
...
  CC      crs_common_all.c
  CC      usart_common_v2.c
  CC      i2c_common_v2.c
../common/i2c_common_v2.c: In function 'i2c_transfer7':
../common/i2c_common_v2.c:431:3: error: 'for' loop initial declarations are only allowed in C99 mode
   for (size_t i = 0; i < rn; i++) {
   ^
../common/i2c_common_v2.c:431:3: note: use option -std=c99 or -std=gnu99 to compile your code
../../Makefile.include:44: recipe for target 'i2c_common_v2.o' failed
make[1]: *** [i2c_common_v2.o] Error 1
Makefile:63: recipe for target 'lib/stm32/f0' failed
make: *** [lib/stm32/f0] Error 2
```

With `CFLAGS="-std=gnu99" make` all build fine, but i think it may be bug in makefile or documentation.

karlp, 2017-04-18:
yes.  I "broke" it recently by using c99 code.  The codebase didn't have any -std= flag, so was getting toolchain defaults, and I've been on a gcc 5.x toolchain for a while now. (which defaults to c11 iirc)

It was reported here, but the "right" thing to do is add the -std=c99 flags to the library makefiles. 

For your particular case, you can also just start using a newer toolchain :)  Sorry about the inconvenience though, it was not intentional to break compiles on older toolchains.

katyo, 2017-04-21:
I have same issue when using GCC ARM toolchain from GNU/Debian

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (4.9.2-10+14~bpo8+1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

karlp, 2017-04-21:
And you have the same solution :)

vdudouyt, 2017-06-09:
I confirm, karlp's solution is working fine for me. Eager to see that in upstream.

closed 2017-08-24 by karlp