libopencm3/libopencm3 #778: Handling USB control requests with wLength > 0

katyo, 2017-04-21:
I developing usb-cdc device and I need to handle SET_LINE_CODING requests which has additional data block.

I found that I can't get this requests.

I wrote simple GDB script to deal with that:
```
b usb_control.c:231 if req->bmRequestType == 0x21
commands 1
p/x req->bRequest
p/x req->wLength
p/x req->wValue
c
end
b usb_control.c:178 if req->bmRequestType == 0x21
commands 2
p/x req->bRequest
c
end
```

To test it I run command stty like that:
```
stty -F /dev/ttyACM0 115200
```

and I get the next output from gdb:
```
Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$10 = 0x22
$11 = 0x0
$12 = 0x3
Breakpoint 2, usb_control_setup_read (usbd_dev=0x20001abc <st_usbfs_dev>, req=0x20001aec <st_usbfs_dev+48>) at ../../usb/usb_control.c:178
178             if (usb_control_request_dispatch(usbd_dev, req)) {
$13 = 0x22

Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$14 = 0x20
$15 = 0x7
$16 = 0x0

Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$17 = 0x22
$18 = 0x0
$19 = 0x0

Breakpoint 2, usb_control_setup_read (usbd_dev=0x20001abc <st_usbfs_dev>, req=0x20001aec <st_usbfs_dev+48>) at ../../usb/usb_control.c:178
178             if (usb_control_request_dispatch(usbd_dev, req)) {
$20 = 0x22
```

As you can see, the SET_CONTROL_LINE_STATE (0x22) requests (with wLength == 0x0) succesfully received and handled, but SET_LINE_CODING (0x20) requests (with wLength == 0x7) received in `_usbd_control_setup` but doesn't catched to `usb_control_setup_read` at all.

karlp, 2017-04-21:
ur doing it wrong :)

Here's an example of handling exactly that control request: https://github.com/karlp/libopencm3-tests/blob/master/tests/usb-serial-rs485/usb_cdcacm.c#L230

Did you actually register a control request callback handler?

closed 2017-04-21 by karlp

katyo, 2017-04-22:
Of course. The fact is that I doesn't get SET_LINE_CONDING request at all.

katyo, 2017-04-22:
I modified gdb script to look `usb_control_request_dispatch` calls:

```
b usb_control.c:231 if req->bmRequestType == 0x21
commands 1
p/x req->bRequest
p/x req->wLength
p/x req->wValue
c
end
b usb_control_request_dispatch if req->bmRequestType == 0x21
commands 2
p/x req->bRequest
c
end
```

The debugger output which I get now:
```
Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$5 = 0x22
$6 = 0x0
$7 = 0x3

Breakpoint 2, usb_control_request_dispatch (usbd_dev=0x20001abc <st_usbfs_dev>, req=0x20001aec <st_usbfs_dev+48>) at ../../usb/usb_control.c:143
143     {
$8 = 0x22

Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$9 = 0x20
$10 = 0x7
$11 = 0x0

Breakpoint 1, _usbd_control_setup (usbd_dev=0x20001abc <st_usbfs_dev>, ea=<optimized out>) at ../../usb/usb_control.c:231
231             if (req->wLength == 0) {
$12 = 0x22
$13 = 0x0
$14 = 0x0

Breakpoint 2, usb_control_request_dispatch (usbd_dev=0x20001abc <st_usbfs_dev>, req=0x20001aec <st_usbfs_dev+48>) at ../../usb/usb_control.c:143
143     {
$15 = 0x22
```

I seems that something went wrong, because I don't see any `usb_control_request_dispatch` calls with bRequest == 0x22 (SET_LINE_CONDING). As I understand the function `usb_control_request_dispatch` must be called in any case.

katyo, 2017-04-22:
Oh, it seems I was a stack-related problem. This issue can be closed.