libopencm3/libopencm3 #779: Race condition in USB control state machine

zyp, 2017-04-21:
The USB control state machine has a race condition in the handling of control reads. If both the data IN packet completes and the status OUT packet arrives between two invocations of poll(), the status OUT packet might be passed to the control state machine first, before it's expected.

This was first reported in #668, ref. [my comment](https://github.com/libopencm3/libopencm3/issues/668#issuecomment-231044266) that describes the problem and my proposed solution. The implemented solution to #668 instead works around the problem by modifying the dwc_otg driver to always process TX events before RX events if both are pending.

During testing of #777 we see the same problem happening on st_usbfs based devices. The DIR bit in USB_ISTR will indicate RX if both events are pending, causing the RX event to be processed first.

To solve this, we could either ignore the DIR bit and manually check CTR_RX and CTR_TX to get the same behavior as the dwc_otg driver, or we could implement the solution that I outlined in the referenced comment that would make the control state machine more robust, not relying on the specific timing of each individual device driver.

jhol, 2017-04-25:
[image]

This image shows the mode of failure. There needs to be a way to either call `_usbd_control_in` before `_usbd_control_out` OR NAK the OUT transaction while allowing the IN transaction to be ACK'd as normal.

jhol, 2017-04-25:
The above trace was created with this test program: https://github.com/jhol/opencm3-usb-error-demo

zyp, 2017-10-26:
This got fixed in #785.

closed 2017-10-26 by zyp