libopencm3/libopencm3 #783: nvic.h not generated during make

theHamClub, 2017-04-26:
As stated in many of the issues, the fix for nvic.h not found is to run make from top level directory which I did but during the generation of headers , nvic.h is not generated in cm3 . 

> root@ubuntu:~/Downloads/fmcw2/firmware/libopencm3# make V=1
  GENHDR  include/libopencm3/stm32/l1/irq.yaml
  GENHDR  include/libopencm3/stm32/f4/irq.yaml
  GENHDR  include/libopencm3/stm32/f3/irq.yaml
  GENHDR  include/libopencm3/stm32/f2/irq.yaml
  GENHDR  include/libopencm3/stm32/f1/irq.yaml
  GENHDR  include/libopencm3/stm32/f0/irq.yaml
  GENHDR  include/libopencm3/sam/3x/irq.yaml
  GENHDR  include/libopencm3/sam/3n/irq.yaml
  GENHDR  include/libopencm3/lpc43xx/m4/irq.yaml
  GENHDR  include/libopencm3/lpc43xx/m0/irq.yaml
  GENHDR  include/libopencm3/lpc17xx/irq.yaml
  GENHDR  include/libopencm3/lpc13xx/irq.yaml
  GENHDR  include/libopencm3/lm3s/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32tg/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32lg/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32gg/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
  BUILD   lib/stm32/f0
make --directory=lib/stm32/f0 SRCLIBDIR=/home/ubuntu/Downloads/fmcw2/firmware/libopencm3/lib
make[1]: Entering directory '/home/ubuntu/Downloads/fmcw2/firmware/libopencm3/lib/stm32/f0'
  CC      vector.c
arm-none-eabi-gcc -Os -g -Wall -Wextra -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -Wundef -Wshadow -I../../../include -fno-common -mcpu=cortex-m0  -mthumb -Wstrict-prototypes -ffunction-sections -fdata-sections -MD -DSTM32F0 -o vector.o -c ../../cm3/vector.c
In file included from ../../cm3/vector.c:21:0:
../../../include/libopencm3/cm3/vector.h:41:33: fatal error: libopencm3/cm3/nvic.h: No such file or directory
 #include <libopencm3/cm3/nvic.h>
                                 ^
compilation terminated.
../../Makefile.include:41: recipe for target 'vector.o' failed
make[1]: *** [vector.o] Error 1
make[1]: Leaving directory '/home/ubuntu/Downloads/fmcw2/firmware/libopencm3/lib/stm32/f0'
Makefile:67: recipe for target 'lib/stm32/f0' failed
make: *** [lib/stm32/f0] Error 2

But nvic.h is generated in the target directories . Should I change the include from libopencm3/cm3/nvic.h to libopecncm3/lpc43xx/m0/nvic.h ?

karlp, 2017-05-08:
I suspec you may have had old .d files lying around perhaps?  Try make clean; make; from the top level directory.  Noone else is seeing this at the moment, so it's hard to say.

karlp, 2017-06-07:
no updates from reporter, nothing more we can do here.

closed 2017-06-07 by karlp

karthikprabhu17, 2019-05-19:
i face this too..any workaround for this?

ChuckM, 2019-05-19:
It looks like problem experienced by @theHamClub originated in part because the fmcw2 project just copied the files from the libopencm3 directory into their repo and created a custom build script for what they wanted to build.