libopencm3/libopencm3 #790: can bit rate timings don't clear old values

karlp, 2017-05-08:
Needs to mask and then or, not just or in the value.  reported on the mailing list.

lib/stm32/can.c around line 176:

        /* Set bit timings. */
        CAN_BTR(canport) |= sjw | ts2 | ts1 |
                ((brp - 1ul) & CAN_BTR_BRP_MASK);

karlp, 2017-05-08:
Are you sure you're seeing this? the can_init() routine clears the entire register here https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/can.c#L129
and then slowly ORs in bit by bit, finally ORing in the timings.

fenugrec, 2017-05-09:
arguing with yourself ? heh. 
The `CAN_BTR(canport) = 0;` line could be moved closer to the |= lines.

ve3wwg, 2017-05-09:
If the registers are cleared, then the values of sjw, ts2 and ts1 are not problematic. I thought was experiencing a problem with the high bit in the brp field, but testing it now, I see it is cleared also. So don't spin your wheels. I would suggest just fixing the CAN_BTR_BRP_MASK macro should be enough, if you want to close this issue.

karlp, 2017-05-09:
@fenugrec it could probably be moved down to https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/can.c#L168 maybe, but shouldn't make any difference, it's all in the same routine.

closed 2017-05-09 by karlp