libopencm3/libopencm3 #804: scripts/genlink.awk: need to make mode a regexp

vollst-induktion, 2017-07-10:
The MODE regexp (given via command line) in "scripts/genlink.awk" was introduced (#240) to filter for specific parameter classes.
If MODE isn't set via command line, MODE was set to ".* ", matching all parameter classes.
On Aug 19, 2016, there was a commit "genlink.awk: no need to make MODE a regexp" by  @karlp
which does not treat MODE as a regexp anymore, i.e.

- All regexp searches "~" were replaced by simple string matches "=="
- MODE is not initialized anymore

Now, if MODE isn't set via command line, the genlink.awk script doesn't work anymore, because NO parameter equals the empty string. This breaks the automated generation of linker scripts as described in "ld/README" and makes the script not working as expected.

The reason for this change was:

> Since I don't understand regular expressions and don't see how they are useful in this case I made the mode selection a simple string match

This commit should be reverted.

[ I'm new to github. Maybe someone will format this and/or link those commits above in a more convenient way. ]

vollst-induktion, 2017-07-10:
As I see right now: Due to the compiler independent notation in "ld/devices.data" the automated generation of linker scripts is broken anyway. "ld/Makefile.linker" should separate the calls to the genlink.awk script using the MODE variable:

`LD_PARAMS=$(shell awk -v PAT="$(DEVICE)" -v MODE="DEFS" ...)`
`LD_PARAMS+=-mcpu=$(shell awk -v PAT="$(DEVICE)" -v MODE="CPU" ...)`
`LD_PARAMS+=-mfpu=$(shell awk -v PAT="$(DEVICE)" -v MODE="FPU" ...)`

karlp, 2017-07-11:
An example of how this is used is https://github.com/libopencm3/libopencm3-examples/blob/master/examples/sam/d/d10-xplained-mini/miniblink/Makefile#L19 leading to https://github.com/libopencm3/libopencm3-examples/blob/master/examples/rules.mk#L92

It appears the docs are wrong, and ld/Makefile.* are probably obsolete.  Please try this alternative out and suggest a PR removing them/rewriting the docs.

vollst-induktion, 2017-07-12:
> Please try this alternative out

Thanks, that works!

> and suggest a PR [...] rewriting the docs.

#805

closed 2017-07-12 by vollst-induktion