libopencm3/libopencm3 #808: The ability to access CAN2 filter registers in dual-can STM32s is meaningless

revzin, 2017-07-21:
On dual-CAN STM32 devices (e. g. STM32F107RB) filter data is shared between CAN2 and CAN1 and is "assigned" to CAN1 (see "Figure 222" in RM0008 or "Figure 335" in RM0090). 

Specifically, every CAN register starting at offset `0x200` exists only in CAN1. There is only one instance of CAN_FMR, CAN_FA1R, CAN_FM1R, CAN_FS1R, CAN_FFA1R, CAN_FiRx on a dual-CAN STM32.

However, libopencm3 provides functions like 
```
void can_filter_init(uint32_t canport, uint32_t nr, bool scale_32bit,
    bool id_list_mode, uint32_t fr1, uint32_t fr2,
    uint32_t fifo, bool enable)
```
and macros like 
```
CAN_FMR(can_base)
```
where `canport`/`can_base` is selectable as in other CAN registers. Calling these functions with `canport = CAN2` has no effect on any CAN filters, confusing the user (spent awhile trying to figure out why nothing is received on CAN2), and actually, I think, tries to write somewhere in BKP registers past bxCAN2.

karlp, 2017-07-21:
Sure, on F107.  But I don't think it would be at all helpful tot have a special case just for those registers.  This is also seen with the ADC and a few other periphs that have shared/common blocks.  The "standard" register style is consistent, but yes, it doesn'tt prevent you from using an invalid argument.

closed 2017-07-21 by karlp

revzin, 2017-07-21:
I do agree that registers are too little of a thing to worry about but I'd consider removing the ``canport`` argument form the five `void can_filter...` functions, what purpose does it have?

karlp, 2017-07-21:
Sure, perhaps, file a PR for that.  You can reference adc_enable_temperature_sensor() as something that also omits the peripheral name.

karlp, 2018-02-14:
merged via it's PR