libopencm3/libopencm3 #822: USB SetAddress

alxsh1979, 2017-09-20:
If the EP0 size is smaller than the usb_device_descriptor size, then the enumeration process stops, because the point does not return to the receive mode until the entire descriptor is finished. If the size of the point exceeds the size of the descriptor, then the host immediately gets it all and sets the address of the device.

[code st_usbfs_core.c]
	if (istr & USB_ISTR_CTR) {
		uint8_t ep = istr & USB_ISTR_EP_ID;
		uint8_t type;

		if (istr & USB_ISTR_DIR) {
			/* OUT or SETUP? */
			if (*USB_EP_REG(ep) & USB_EP_SETUP) {
				type = USB_TRANSACTION_SETUP;
			} else {
				type = USB_TRANSACTION_OUT;
			}
		} else {
			type = USB_TRANSACTION_IN;
			USB_CLR_EP_TX_CTR(ep);
		}

		if (dev->user_callback_ctr[ep][type]) {
			dev->user_callback_ctr[ep][type] (dev, ep);
		} else {
			USB_CLR_EP_RX_CTR(ep);
		}
	}
[/code]

Change:
		if (dev->user_callback_ctr[ep][type]) {
			dev->user_callback_ctr[ep][type] (dev, ep);
		}
		USB_CLR_EP_RX_CTR(ep);
	}

karlp, 2017-09-21:
Could you please clarify what you mean by EP0 size?  If you mean the size of the control buffer, then yes, that does need to be at least as big as your descriptor.  If you mean something else, I'm not sure what you're describing.

alxsh1979, 2017-09-21:
The size of the buffer EP0 and .bMaxPacketSize0 can be less than the size of the usbDeviceDescriptor. In my case, .bMaxPacketSize0 = 8. The size of the descriptor is 18 bytes. At the beginning of enumeration, the host requests usbDeviceDescriptor. The controller sends it an initial 8 bytes. The host does not request a remainder and immediately assigns the device address. After that, the host again requests usbDeviceDescriptor completely. It's according to the standard.

karlp, 2017-09-21:
Ok, yes, it's completely ok that bMaxPacketSize0 is less than tthe descriptor, that's no problem.  Do you have some test case where this is happening?  You could modify the gadget-zero test cases if you just want to hack something up.  We've got a lot of cases where bMaxPacketSize0 is less (often much less) than the descriptor size and we're not seeing any problems with them.

alxsh1979, 2017-09-22:
The libopencm3 code in the USB part is ported to the IAR. I attach the archive file with important files. They will have to be fixed for compilation in GCC. Target platform STM32F042C6T6.
[usb.ZIP](https://github.com/libopencm3/libopencm3/files/1323717/usb.ZIP)

karlp, 2017-09-22:
a) please reproduce this with libopencm3 unmodified.  I'm not very interestedin fixing bugs in someone elses porting.
b) what are you doing with the strings?!  Have you looked at existing code? or are you trying to port line by line from somewhere else?
c) What is the actual _problem_ that you're getting?

karlp, 2017-10-26:
Please feel free to re-open this issue if you can reproduce it with plain libopencm3 code, without further information here, there's nothing useful we can do.

closed 2017-10-26 by karlp