libopencm3/libopencm3 #826: Compiler warning with gcc 7.2.0

anatol, 2017-09-28:


karlp, 2017-09-28:
do they suggest a replacement?  Because that seems pretty plain usage to me.

karlp, 2017-09-28:
Can you try this out please? https://github.com/libopencm3/libopencm3/pull/827

anatol, 2017-10-06:
Thanks a lot. This warning has gone. BTW that are few more warnings that you might want to looks at:

```
../can.c: In function 'can_transmit':
../can.c:408:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdhxr.data8[3] = data[7];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:410:2: note: here
  case 7:
  ^~~~
../can.c:411:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdhxr.data8[2] = data[6];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:413:2: note: here
  case 6:
  ^~~~
../can.c:414:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdhxr.data8[1] = data[5];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:416:2: note: here
  case 5:
  ^~~~
../can.c:417:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdhxr.data8[0] = data[4];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:419:2: note: here
  case 4:
  ^~~~
../can.c:420:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdlxr.data8[3] = data[3];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:422:2: note: here
  case 3:
  ^~~~
../can.c:423:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdlxr.data8[2] = data[2];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:425:2: note: here
  case 2:
  ^~~~
../can.c:426:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
   tdlxr.data8[1] = data[1];
   ~~~~~~~~~~~~~~~^~~~~~~~~
../can.c:428:2: note: here
  case 1:
  ^~~~
```

karlp, 2017-10-06:
(further, did you actually _look_ at the code?  it's very explicit, it just didn't use gcc's magical syntax to tell it the _right_ way)

closed 2017-10-06 by karlp

anatol, 2017-10-10:
Thank you Karl for fixing it. Now the build is clean of warnings.