libopencm3/libopencm3 #828: Compilation error

adichell, 2017-09-29:
Hi,

I downloaded the libopencm3-master, and with the following PATH values I tried to compile the library:

`C:\msys\1.0\bin;C:\Python27;C:\gcc-arm-none-eabi-5_4_2016q3\bin;`

I get the following error:

`~\libopencm3-master>make
FIND: Parameter format not correct
File not found - *.[ch]
  BUILD   lib/stm32/f0
  CP      libopencm3_stm32f0.ld
  CC      can.c
  CC      flash.c
  CC      rcc.c
  CC      usart.c
  CC      dma.c
  CC      rtc.c
  CC      comparator.c
  CC      crc.c
  CC      dac.c
  CC      iwdg.c
  CC      pwr.c
  CC      gpio.c
  CC      timer.c
  CC      adc.c
  CC      desig.c
  CC      gpio_common_all.c
  CC      gpio_common_f0234.c
  CC      crc_common_all.c
  CC      pwr_common_v1.c
  CC      iwdg_common_all.c
  CC      rtc_common_l1f024.c
  CC      dma_common_l1f013.c
  CC      exti_common_all.c
  CC      spi_common_all.c
  CC      spi_common_f03.c
  CC      flash_common_f01.c
  CC      dac_common_all.c
  CC      timer_common_all.c
  CC      timer_common_f0234.c
  CC      rcc_common_all.c
  CC      adc_common_v2.c
  CC      crs_common_all.c
  CC      usart_common_v2.c
  CC      i2c_common_v2.c
  CC      usb.c
  CC      usb_control.c
  CC      usb_standard.c
  CC      st_usbfs_core.c
  CC      st_usbfs_v2.c
  CC      vector.c
In file included from ../../../include/libopencm3/cm3/nvic.h:133:0,
                 from ../../../include/libopencm3/cm3/vector.h:41,
                 from ../../cm3/vector.c:22:
../../../include/libopencm3/dispatch/nvic.h:6:39: fatal error: libopencm3/stm32/f0/nvic.h: No such file or directory
compilation terminated.
make[1]: *** [vector.o] Error 1
make: *** [lib/stm32/f0] Error 2
`

Best regards,
A.

karlp, 2017-09-29:
run make from a shell with the right find.  (did you get a shell with msys?)

closed 2017-09-29 by karlp

karlp, 2017-09-29:
Alternatively, if you can test this patch works on windows for me, I can merge it.  I know it works on linux, but I don't have a windows pc around to test it on

adichell, 2017-09-29:
You are right, using the msys shell compilation under windows does work. Thanks!

karlp, 2017-10-18:
I've merged this anyway, it's at least as workable as before, and hopefully better, absent feedback.