libopencm3/libopencm3 #831: Issue with DNF field in i2c_set_digital_filter

davidgfnet, 2017-09-30:
It seems that by default the DNF definitions are shifted by 8 (to move them to the right position), for example #define I2C_CR1_DNF_UP_2_TI2CCLK    (0x2 << 8)

Now the function that sets it:

````
void i2c_set_digital_filter(uint32_t i2c, uint8_t dnf_setting)
{
    I2C_CR1(i2c) = (I2C_CR1(i2c) & ~I2C_CR1_DNF_MASK) | dnf_setting;
}
````

As we can see the param is a byte, therefore truncates the constant. So I'd change the defines to not include a shift and then add a shift in the function body. Or we could well just use uint16_t or bigger.

Thanks for the work to all the authors! :)

ChuckM, 2017-09-30:
In include files and code I've created I have typically stuck with the idiom:
```
    reg = (PARAMETER << REG_PARAMETER_SHIFT) | /* other parameters */;
    /* or for replacement */
   reg = (reg & ~(PARAMETER_MASK << PARAMETER_SHIFT) | (PARAMETER << PARAMETER_SHIFT);
```
Using that idiom, this would be
```
void i2c_set_digital_filter(uint32_t i2c, uint8_t dnf_setting)
{
    I2C_CR1(i2c) = (I2C_CR1(i2c) & ~(I2C_CR1_DNF_MASK << I2C_CR1_DNF_SHIFT)) | (dnf_setting << I2C_CR1_DNF_SHIFT);
}
```

To 'fix' this code, I would change `I2C_CR1_DNF_UP_2_TI2CCLK` to have the value 2 and add

```
#define I2C_CR1_DNF_SHIFT 8
#define I2C_CR1_DNF_MASK 8
```

As has been mentioned in other PRs if you make the mask and shift defines algorithmic like this it is easier to write a macro which always does the correct shifts.

karlp, 2017-10-26:
Do you have some code that uses this?  Could you verify a patch?

closed 2018-01-08 by karlp