libopencm3/libopencm3 #858: nvic_set_priority needs fix on STM32F0 Cortex M0

hubmartin, 2017-11-30:
Hi, I'm porting Black Magic Probe which is using libopencm3 to the STM32F0 which is Cortex M0. I wasn't able to set priority levels using nvic_set_priority, later I've noticed that #332 mentioned, that NVIC PRI registers are only word-accessible. Just would like to point this out so others can find this issue later. Not sure if and when I publish some pull request.

Funny thing was that I was able to change number in the debugger in registers, but not by the code even when I was writing to right address the right value :)

karlp, 2017-11-30:
closing in favour of the already existing #332

closed 2017-11-30 by karlp

hubmartin, 2017-11-30:
The original Issue was related to the bit shift for CM3. My issue is about completely different bug which is related to the word access to NVIC PRI registers on STM32F0. Just wanted to point that out.