libopencm3/libopencm3 #859: i can't trigger sys_tick_handler

alphasion, 2017-12-03:
latest code(953bf53),
`#include "libopencm3/stm32/rcc.h"
#include "libopencm3/stm32/flash.h"
#include "libopencm3/stm32/gpio.h"
#include "libopencm3/cm3/systick.h"
#include "libopencm3/cm3/nvic.h"
static const struct rcc_clock_scale clock_setup =
{ .pllm = 8, .plln = 336, .pllp = 2, .pllq = 7,
		.hpre = RCC_CFGR_HPRE_DIV_NONE, .ppre1 = RCC_CFGR_PPRE_DIV_4, .ppre2 = RCC_CFGR_PPRE_DIV_2,
		.power_save = 0, .flash_config = FLASH_ACR_ICE | FLASH_ACR_DCE | FLASH_ACR_LATENCY_5WS,
		.apb1_frequency = 42000000, .apb2_frequency = 84000000, };
volatile uint32_t system_millis = 0;
void sys_tick_handler(void)
{
	system_millis++;
}

static void msleep(uint32_t delay)
{
	uint32_t wake = system_millis + delay;
	while (wake > system_millis)
		;
}

int main(void)
{
	rcc_clock_setup_hse_3v3(&clock_setup);
	rcc_periph_clock_enable(RCC_GPIOG);
	gpio_mode_setup(GPIOG, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, GPIO13);
	systick_set_clocksource(STK_CSR_CLKSOURCE_AHB);
	systick_set_reload(168000);
	systick_counter_enable();
	systick_interrupt_enable();
	gpio_set(GPIOG, GPIO13);
	while (1)
	{
		gpio_toggle(GPIOG, GPIO13);
		msleep(500);
	}
}`
but sys_tick_handler never triggered....

karlp, 2017-12-03:
What target hardware are you working with here?

karlp, 2017-12-03:
I took your code paste from https://pastebin.com/raw/aZJyVzJJ which is the same as this as far as I can tell, changed GPIOG to GPIOD and GPIO13 to GPIO12, and it compiles, flashes and blinks happily on the f4 discovery.   Do other examples work for you?  It seems like something more basic is missing.  I didn't try using your linker script, I just used the one from https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f4/stm32f4-discovery/fancyblink/Makefile#L22

alphasion, 2017-12-04:
ok,i'll get another try

adamheinrich, 2017-12-26:
Hi @alphasion, what toolchain are you using? I started having the same issue after an update to:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Older versions of the GNU ARM Embedded Toolchain work fine.

adamheinrich, 2017-12-26:
Looks like `-flto` when linking the binary causes the problem. The interrupt handler is included correctly without `-flto`.

karlp, 2017-12-26:
I'll close this in the new year without further information.  The examples work, I can only presume you've copied something badly

closed 2018-01-08 by karlp

alphasion, 2018-01-11:
seems sth wrong with my toolchain, i fixed it already.thanks

karlp, 2018-01-11:
great, thanks for confirming