libopencm3/libopencm3 #870: Unclear sentence in README.md

PeterMortensen, 2018-01-07:
This is about this sentence in README.md:

  "NOTE We recommend, that you use g-a-c version 2.8 2014q3 or newer to build all platforms covered by libopencm3 successfully."

It is not at all clear what "g-a-c" is. Does it refer to "gcc-arm-embedded" (but should it then be "g-a-e"? - what is the "c" for)? To gcc (but the version number does not fit at all)? Or something else?

A 2014q3 version of something is announced in "Rlease GCC ARM Embedded 4.8 update 3", https://launchpad.net/gcc-arm-embedded/+announcement/12983. But the version number is 4.8, not 2.8.

I suggest making this more clear.

karlp, 2018-01-08:
did you cover this already in your other patch?  Yes, this should have been g-a-e, not g-a-c. and yes, 4.8

karlp, 2018-02-14:
Pulled this into my staging branch

closed 2018-02-24 by karlp