libopencm3/libopencm3 #876: SPI examples broken for F1?

Peque, 2018-02-08:
I tried to configure SPI as in the [libopencm3-example](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f1) template:

```c
static void setup_spi(void)
{
	gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_50_MHZ,
		      GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO12 | GPIO13 | GPIO15);
	gpio_set_mode(GPIOB, GPIO_MODE_INPUT, GPIO_CNF_INPUT_FLOAT, GPIO14);

	spi_reset(SPI2);

	spi_init_master(SPI2, SPI_CR1_BAUDRATE_FPCLK_DIV_128,
			SPI_CR1_CPOL_CLK_TO_0_WHEN_IDLE,
			SPI_CR1_CPHA_CLK_TRANSITION_1, SPI_CR1_DFF_8BIT,
			SPI_CR1_MSBFIRST);

	spi_enable_software_slave_management(SPI2);
	spi_set_nss_high(SPI2);

	spi_enable(SPI2);
}
```

It the example, they are not setting/clearing the NSS pin, so I guessed that was done automatically when writing/reading, but when I tried I saw the NSS signal was not changed at all (D0, in black, would be the NSS signal):

[image]

So I thought maybe I had to manually set/clear that pin before reading/writing, but in order to do that, I had to modify the setup (i.e.: just set different GPIO mode for the NSS pin):

```c
static void setup_spi(void)
{
	gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_50_MHZ,
		      GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO13 | GPIO15);
	gpio_set_mode(GPIOB, GPIO_MODE_OUTPUT_50_MHZ, GPIO_CNF_OUTPUT_PUSHPULL,
		      GPIO12);
	gpio_set_mode(GPIOB, GPIO_MODE_INPUT, GPIO_CNF_INPUT_FLOAT, GPIO14);

	spi_reset(SPI2);

	spi_init_master(SPI2, SPI_CR1_BAUDRATE_FPCLK_DIV_128,
			SPI_CR1_CPOL_CLK_TO_0_WHEN_IDLE,
			SPI_CR1_CPHA_CLK_TRANSITION_1, SPI_CR1_DFF_8BIT,
			SPI_CR1_MSBFIRST);

	spi_enable_software_slave_management(SPI2);
	spi_set_nss_high(SPI2);

	spi_enable(SPI2);
}
```

Then I could manually set/clear the NSS pin and everything seemed to work fine:

[image]

I wonder if the example could be broken (i.e.: the user needs to manually clear/set NSS and needs to change the GPIO configuration). Or if I missed something and there is no need to manually set/clear that pin (in that case I do not know what do I have to change in the SPI setup).

In case there is something broken... I have seem the same template for other STM32Fx too in libopencm3-examples, could they be broken too or could this be just the F1? (unfortunately I cannot try with any other STM32 family)

markus-k, 2018-04-05:
I have noticed the same thing. The [example](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32vl-discovery/spi/spi.c) just seems to send data and read it back without using the chip select at all. The NSS pin is not driven low because the SPI is configured for software slave management, and the GPIO output is disconnected when the pin is in AF mode. 

It's probably a good idea to add an example that actually does something more useful, it certainly caused some head scratches for me when I noticed it. This issue is probably better kept in the [examples](https://github.com/libopencm3/libopencm3-examples/issues) repo though.

Peque, 2018-04-18:
@markus-k Yeah, I put it here because I did not know if it was an issue with the examples repository or if it could reallly be an issue with the implementation of the SPI configuration functions (perhaps they should be doing something else).

@karlp Any thoughts on this? :innocent:

alphalove, 2019-04-23:
Hello,
Old I know but... I had a similar problem.  I believe the 'example template code' expects there to be an external pull-up resistor on the NSS pin, without this the code doesn't work.  I had a breadboard version working with an W25Q flash chip, but forgot the pull-up on the PCB version I made (using an AT25D) but it didn't work as expected.  As you stated, you have to resort to driving the NSS pin as per your modified code.

The pull-up (and a bit about the issues with NSS) is explained in the excellent 'Beginning STM32' book by Warren Gay.  This is a must have if you are using STM32 and libopenCM3.

Peque, 2019-10-24:
@alphalove In the case I posted above the peripheral (a gyroscope) is mounted on a PCB that already has a 10k pull-up resistor on NSS. You can [find the board on Aliexpress](https://www.aliexpress.com/item/32880687289.html) (or by simply searching for "MPU 6500" PCBs).

So maybe there is still something wrong software/configuration-side?