libopencm3/libopencm3 #877: Make Install does not work

danieljmg, 2018-02-13:
Hello,
If I want to generally install it in my Ubuntu 17.10 machine, the Install action generates a: "No Install directives found". I don't know if it is related with and 'Ignore All' warning that Make action generates for each folder/device (f1,f2,f3,f4...).
In the documentation 'make install' is mentioned, however, it does not work.

karlp, 2018-02-13:
Which documentation talks about "make install" please?  I can't find any, and the make install target was dropped back in 2015 via 7c1cd2e5cd3f3d1193ee375b46cfd8afaccb89f7

danieljmg, 2018-02-15:
Oh! I have probably mixed readmes from blogs, tutorials, forks. I'm sorry about it. I've managed to do it the right way though.

closed 2018-02-15 by danieljmg