libopencm3/libopencm3 #896: Support for PlatformIO

jcw, 2018-03-18:
The current support for libopencm3 in PlatformIO is limited - see this list of boards:
https://platformio.org/boards?filter%5Bframeworks%5D=libopencm3

As I understand it, PIO works off a clone in `~/.platform/packages/framework-libopencm3/`, although right now, that seems to be an older version. For starters, I'd limit the first steps to just STM32 - as an example, see [this board file](https://github.com/platformio/platform-ststm32/blob/develop/boards/nucleo_l011k4.json) in PIO. The way to get `libopencm3` supported for this Nucleo-L011K4 example is to add "libopencm3" in the "frameworks" list, but when I tried I ran into some puzzling error messages.

I'm still feeling my way around as novice in both PIO and LOCM3, so it's not clear to me what _exactly_ needs to be done to improve the situation between these two worlds.

For completeness:
* a reference back to PR #895 
* a summary of my first steps on the PIO side: https://github.com/platformio/builder-framework-libopencm3/issues/1

Sooo... assuming there's interest: how to go about supporting a _lot_ more boards in PIO?
I'm using the Nucleo L011 as example, since I have one to test with, but it's just one of many candidates.

karlp, 2018-03-18:
well, let us know what you need from this side.

jcw, 2018-03-19:
I'm stuck right now, so if no-one has a suggestion on how to move forward, I'll have to let this go.
Maybe @ivankravets has a suggestion.

karlp, 2018-03-19:
sorry, I've only used platformio for an esp8266 project, I haven't even tried using it for libopencm3.

ivankravets, 2018-03-22:
I think we can move https://github.com/platformio/builder-framework-libopencm3/blob/master/libopencm3.py to this repo and maintain it together. We did the same for ESP8266 / ESP32:

* https://github.com/esp8266/Arduino/blob/master/tools/platformio-build.py
* https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py

karlp, 2018-03-22:
I don't really care where the file's kept, I simply don't understand what the actual _issue_ is.  

PIO is just pointing to a built libopencm3, and using the libopencm3 linker script generator to support all known devices right?  

(trick question, I can see that you've actually written a scons build that parses the makefile itself (wat) to try and rebuild some portion of it, and then only support the few boards that got a custom ld script built into the repo)

I'm not sure what's really required from our side here to "enable" anything in PIO.

ivankravets, 2018-03-22:
PIO is based on http://scons.org/ build tool. All PIO-based dev/platforms depend on PIO Build System API. When PIO starts building a project, it provides all information about hardware/board, packages and etc. PIO does all this work automatically.

The only one moment which we should resolve in this script is to provide a list of files intended for the building. That is why parse makefile to find includes and build files.

> I'm not sure what's really required from our side here to "enable" anything in PIO.

As I understand, we need to extend this script to support other boards.

jcw, 2018-03-22:
As I understand it, libopencm3 now generates linker scripts from this datafile:

https://github.com/libopencm3/libopencm3/blob/master/ld/devices.data

All the generic stuff is now a single `linker.ld.S` file.

jcw, 2018-03-22:
> we need to extend this script to support other boards

Board support will always be a moving target. See Karl's [miniblink](https://github.com/karlp/libopencm3-miniblink) repository, where a whole slew of boards are defined - very minimally - to be able to generate a small blink demo, e.g. `boards.stm32.mk`.

I suspect that PIO is going to run into this combinatorial explosion as well sooner or later - maybe some of these techniques can be applied to keep a handle on things? The combinations of frameworks, µC vendors, boards, is already becoming somewhat of a hit-and-miss chase: pick a board, oh wait, it doesn't support that framework + library I need, pick a smaller version µC which is compatible, oh wait, there's no framework support for that one, etc ...

ivankravets, 2018-03-22:
I added this issue to our TODO list. I none helps us here, we will find own time for that.

karlp, 2018-03-22:
Yes, we provide a variety of ways that people can use linker scripts depending on what their goals are.  They can 
a) provide completely their own script, taking care to use variable names that match our startup code
b) provide just a "memory section" and include the generic base file.  (these are tradtionall called "libopencm3_[family].ld, but see https://github.com/libopencm3/libopencm3/pull/892 for where this will become simpler, and hopefully easier and more obvious)
c) use the dynamic linker script generator.  This is currently implemented as some makefile stubs that you include in your own makefile. but see https://github.com/libopencm3/libopencm3/pull/856 for conversion of the linker script generator from not very portable AWK to python.  This is staged locally, and will land soon, likely within the month.

As for the rest of your build tools, I really recommend that you just build libopencm3 as the library that it ships as, and just provide -L, -I and -D flags as you do for esp8266/esp32, rather than trying to interpret the makefile itself and rebuild portions of it.  We make _zero_ claims that the makefile structure will stay stable, only that the libraries built will be the same.

I personally have zero time to work on your scons files.  I'll leave this issue open here incase someone else in the community wants to work on it.

jcw, 2018-04-29:
This is great - it'd be nice if PIO could fully take advantage of this, now that it's in Python and much closer to the way PIO itself is implemented.

The combinatorial explosion of frameworks x µC archictectures x boards is starting to show: so many boards are _not_ (yet) supported in PIO, even though all the pieces are usually (almost) there. This is not some vague desire of mine "to support everything", it's really because often you want to use a specific library, and have to adjust your choice of framework to be able to do so.

ivankravets, 2018-05-03:
@karlp Thanks! Could you give us some recommendations how to better automate build process of this library?

karlp, 2018-05-03:
@ivankravets I have no idea what you're asking of me.  _this_ project is a library, it has a build system that builds all the lib*.a files.  you do that once.  Then you treat it like any other library, and pass appropriate -I -l and -L flags.

ivankravets, 2018-05-03:
No no, I mean the "logic" what include/exclude for building depending on the target. The same with LD script. I see you mentioned above about some script.

jcw, 2018-05-03:
Ok, so libopencm3 is a framework/runtime and platformio is a build environment. It ought to make for a nice and happy marriage, and a long, prosperous, and synergetic time together.

AIUI, libopencm3 has addressed the issue of dealing with a ton of µC variants, getting all the .ld files right. While I'm clueless w.r.t. the "how", my hope is that PIO can leverage this, so that the `framework = libopencm3` setting in `platformio.ini` can be used for as many boards and µC's out there as possible.

Is there any overlap? I don't think so. PIO exists to brings toolchain(s) and frameworks such as libopencm3 together, right?

(I'm not asking for anyone to put in effort, it's all OSS and we're all in this as volunteers after all, but it would be nice if this issue offers at least some idea what it would take to make this happen, eventually)

funlw65, 2018-05-03:
https://platformio.org/pricing

karlp, 2018-05-03:
unless I'm misunderstanding something grossly, still,  platformio uses scons to build a pile of user provided source files, and has knowledge of what user selected framework they want to use, and what user selected device they want to target.

the target information can be given to https://github.com/libopencm3/libopencm3/blob/master/scripts/genlink.py to generate a linker script.  I know that scons can declare external builders.

the target information can also be used to determine which cross compiler target/arch flags need to be chosen (presumably it's already getting this right)  The same information can be used to determine whether it should pass "-lopencm3_stm32f3" or "-lopencm3_samd"  and likewise wether to pass "-DSTM32F7" or "-DEFM32_HG"

That map of target/define/library name is not kept in a json file, it's kept in individual project makefiles, as it's end user specific, though you can help prompt people.

the boards.*.mk files in the https://github.com/karlp/libopencm3-miniblink repository are probably one of the more compact representations of the flags needed, though many of these are just the cortex-m0 vs m0+ vs m3 vs m4(f) flags that platformio would already have know how to get right.

Hopefully, there's sufficient information there to help you write a scons file that that links against an existing library.

karlp, 2018-05-04:
I'm going to close this ticket, it's probably not productive to keep it as is.  If anyone has specific issues they'd like addressed that would make it easier to integrate with any other project, feel very welcome to open new issues or PRs.

closed 2018-05-04 by karlp

ivankravets, 2018-05-04:
@karlp thank you so much for the detailed explanation. I've just updated our feature request https://github.com/platformio/builder-framework-libopencm3/issues/1