libopencm3/libopencm3 #914: No support for DMA in STM32L0

MarSik, 2018-04-29:
There are no includes with DMA functions for STM32L0.

karlp, 2018-04-29:
there's lots of peripherals on lots of hardware that isn't supported, we cannot have a ticket for every peripheral on every single target, that wouldn't even be helpful to track anything.  DMA on l0 probably needs very little testing and adding in the common dma code.  Feel free to open a PR for that, but I'm not going to keep broad tickets like this open.

closed 2018-04-29 by karlp

MarSik, 2018-04-29:
How do you track missing peripherals then? Especially when the other platforms have the support ready? DMA is pretty core component for low power designs.. which is what L0 is about.

karlp, 2018-04-29:
I don't.  You can edit and maintain pages on the wiki if you like, but it's an endless task, and never descriptive enough for people.