libopencm3/libopencm3 #917: Hard fault in reset_handler

tagunil, 2018-05-06:
GCC 7.2 emits following code for reset_handler:
```
   0x080000c8 <+0>:     movs    r2, #0
   0x080000ca <+2>:     ldr     r4, [pc, #140]  ; (0x8000158 <reset_handler+144>)
   0x080000cc <+4>:     ldr     r3, [pc, #140]  ; (0x800015c <reset_handler+148>)
   0x080000ce <+6>:     add     r4, pc
   0x080000d0 <+8>:     ldr     r3, [r4, r3]
=> 0x080000d2 <+10>:    str     r3, [sp, #4]
   0x080000d4 <+12>:    ldr     r3, [pc, #136]  ; (0x8000160 <reset_handler+152>)
   0x080000d6 <+14>:    ldr     r3, [r4, r3]
   0x080000d8 <+16>:    str     r3, [sp, #8]
   0x080000da <+18>:    ldr     r3, [pc, #136]  ; (0x8000164 <reset_handler+156>)
   0x080000dc <+20>:    ldr     r3, [r4, r3]
   0x080000de <+22>:    str     r3, [sp, #12]
   0x080000e0 <+24>:    ldr     r3, [sp, #4]
   0x080000e2 <+26>:    ldr     r0, [sp, #12]
```
Local variables are written here without decrementing stack pointer first, so it is an out-of-SRAM write.

reset_handler is marked as naked. GCC manual says about naked functions:
> Only basic asm statements can safely be included in naked functions (see Basic Asm). While using extended asm or a mixture of basic asm and C code may appear to work, they cannot be depended upon to work reliably and are not supported.

tagunil, 2018-05-08:
Well, I've discovered that my GCC build is quite unusual and does strange things all over the place. Nevertheless, naked attribute should not be used here.

karlp, 2018-05-08:
yes, you're right, we've actually got it lying around with a few other changes to make it ~possible to build with clang.  But it's not critical right now, as this has been perfectly functional with recommended toolchains for years.

07151129, 2018-05-18:
You should also use the interrupt attribute for interrupt handler routines to ensure AAPCS compatibility, as well as compatibility with other Cortex CPUs (just in case).

karlp, 2018-07-29:
this was merged via f53e12d2

closed 2018-07-29 by karlp