libopencm3/libopencm3 #918: USART overrun prevents reception

jsphuebner, 2018-05-09:
Hi,

I experienced frequent uart lockups on an stm32f103 when the system was busy with various ISRs. I'm currently doing synchronous reception using usart_recv_blocking().
It turns out RXNE is no longer set after an overrun, i.e. ORE. This doesn't quite correspond to the manual, but hey.
So the culprit is here:
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/common/usart_common_f124.c : 95

while ((USART_SR(usart) & USART_SR_RXNE) == 0);

Should be 
while ((USART_SR(usart) & USART_SR_RXNE) == 0 && (USART_SR(usart) & USART_SR_ORE) == 0);

See also:
https://github.com/ARMmbed/mbed-os/issues/1605

karlp, 2018-05-09:
I highly recommend you not actually try using functions like usart_recv_blocking() for anything more than demonstrations.  The fix you link is exactly how any sane irq handler would be written.

jsphuebner, 2018-05-09:
I will move to DMA. That said usart_recv_blocking was handy in my application as the main() is used only for the terminal. All "real" work is done in various interrupt handlers.
Any reason not to change usart_recv_blocking() as I suggested?
Thanks.

karlp, 2020-04-09:
I don't believe the usart receive should be checking for error flags that shouldn't be set if you're running things normally. Even the referenced mbed issue is fixing this is a proper irq handler, where you would not be using this function.

closed 2020-04-09 by karlp

jsphuebner, 2020-04-16:
I don't think it makes sense to have functions that loop indefinitely. If you don't want people to use the function, delete it rather than leaving it at a defunct state that takes ages to debug.