libopencm3/libopencm3 #921: scb_reset functions need to issue a sync barrier

07151129, 2018-05-18:
As per ARM AN321, you should issue a DSB right after writing to AIRCR to make sure the writes propagate before the reset. It is also probably a good idea to mask IRQs, as user code may not expect IRQ handlers to execute during a reset and be interrupted, possibly causing some damage.

ChuckM, 2018-05-18:
Masking IRQs is unnecessary as the Reset will occur at the next instruction fetch cycle (so no exceptions can be serviced before it hits) the easiest way to prove that this is true is to create a hard fault condition on the instruction just after the SCB write, it never executes.

That said, insuring that everything has been written to memory *before* the reset is a solid request.

07151129, 2018-05-18:
By this reasoning you could also argue that the loop after writing to AIRCR is not necessary. In fact, the AN states that DSB can be omitted unless CPSID is used. I still believe it is a good idea to have both DSB and CPSID to account for possible behavioural differences between Cortex M implementations, and not to deviate from CMSIS.

07151129, 2018-05-18:
I should also mention that reset should preserve PRIGROUP, as explained by ARM: https://github.com/ARM-software/CMSIS_5/issues/302