libopencm3/libopencm3 #941: gadget0 code does not re-enumerate on reset

vampirefrog, 2018-07-08:
Target: STM32F407 (using usb-gadget0-stm32f4disco.elf). When pressing reset or doing 'run' from gdb, on latest revision of libopencm3, I do not see the dmesg events that would show the device info. I can only compare this to unicore-mx examples, because I don't know of a real-world device with a reset button. But it seems that there is a need to generate some sort of fake reconnect, so that the OS re-reads the descriptors and re-displays them in dmesg.

```
[  535.754584] usb 4-3: new full-speed USB device number 7 using ohci-pci
[  535.937357] usb 4-3: New USB device found, idVendor=cafe, idProduct=cafe
[  535.937365] usb 4-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  535.937370] usb 4-3: Product: Gadget-Zero
[  535.937375] usb 4-3: Manufacturer: libopencm3
[  535.937379] usb 4-3: SerialNumber: stm32f4disco
```

vampirefrog, 2018-07-08:
Sorry, this is embarrasing. The code works fine on an actual STM32F4-Discovery board, I was using one of [these bad bois](https://www.aliexpress.com/item/STM32F407VET6-development-board-M4-STM32F4-core-board-arm-development-board-cortex-M4/32834250667.html). It has an external 1.5k resistor to 3V on DP, and I set it to ground then to 3V for a few cycles and it reconnects fine. Is there a way to disable the internal pullup? Thx.

karlp, 2018-07-09:
yes.  I told you it worked perfectly well on the actual hardware :)  if you buy random hw, you need to be aware of that.   something that might work for you, depending on how it's wired is to follow what the f1 code does, overriding teh gpio briefly: https://github.com/libopencm3/libopencm3/blob/master/tests/gadget-zero/main-stm32f103-generic.c#L46-L56

closed 2018-07-09 by karlp

vampirefrog, 2018-07-09:
Thanks, that's what I did.