libopencm3/libopencm3 #944: Cleaner/better heap support

ChuckM, 2018-07-13:
This is posted primarily for discussion.

The heap allocator in newlib is reasonable for small systems. It works fairly simply by starting with a single "4k" block and when it runs out it calls sbrk() to get more memory.  That ends up calling the weak symbol _sbrk_r which can be overridden in user code. (the default returns -1, there is no more memory).

If we define the end of allocated memory in the generic linker script like so:
```
    .heap : {
        . = ALIGN(4);
        _heap_start = .;
    } >ram
```
Then we can write our own version of `_sbrk_r` that works like this:
```
/*
 * sbrk.c
 *
 * This implements a larger heap than you get by default
 * in newlib. It uses the RAM between the end of the declared
 * data and the bottom of the "largest" stack for the heap.
 */
#include <stdint.h>
#include <stdio.h>
#include <errno.h>
#include <malloc.h>
#include "../util/util.h"

#define MAX_STACK_SIZE  8192

/* these are defined by the linker script */
extern uint8_t _heap_start, _stack;

static uint8_t *_cur_brk = &_heap_start;

/* prototype to make gcc happy */
void *_sbrk_r(struct _reent *, ptrdiff_t );

void *_sbrk_r(struct _reent *reent, ptrdiff_t diff)
{
    uint8_t *_old_brk = _cur_brk;
    if (_cur_brk + diff > (uint8_t *)(&_stack - MAX_STACK_SIZE)) {
        reent->_errno = ENOMEM;
        return (void *)-1;
    }
    _cur_brk += diff;
    return _old_brk;
}
```
Now `malloc` and `calloc` will work as expected up until you run out of memory. Note that if for some reason there isn't enough ram to allow any heap then the behavior goes back to the current behavior which is that newlib fails to allocate any memory past the 4K it starts with.

I would like to suggest we put `_heap_start` into the generic LD script so that this code can work. And while I don't think everyone will use it it would be possible to add it to the readme/documentation and help out folks who were trying to use these functions.

karlp, 2018-07-13:
is this not just _ebss ? do you really need anything new for this?

ChuckM, 2018-07-13:
Hmm, looking more closely at it, you are probably correct. Also it makes it a bit "hidden" in the linker script in the sense that one has to just "know" that is where the heap will start. I can work with that though.

I was thinking there was no way to put an `sbrk_r` implementation into the library although I believe I have found a workaround for that as well. I will post it here in a sec after some more testing.

ChuckM, 2018-07-14:
ok, here is a variation and it uses `_ebss` 
```
/*
 * sbrk.c
 *
 * This implements a larger heap than you get by default
 * in newlib. It uses the RAM between the end of the declared
 * data and the bottom of the "largest" stack for the heap.
 *
 * The only user interface to this code is the weakly bound
 * local_heap_setup() which is called the first time malloc
 * has to get more memory for the heap. This function passes
 * pointers to its two controls, the start address and end
 * address of the heap. Once called those are not changed.
 *
 * If the user does not define local_heap_setup in their code,
 * the RAM between _ebss and _stack - MAX_STACK_SIZE is used
 * for the heap. Typically this is RAM that is internal to the
 * SoC.
 *
 * On other processors, additional ram might become available
 * after reset and some peripheral setup (like PSRAM or SDRAM
 * on parts with the FMC peripheral). In that case the user
 * can define their own function local_heap_setup and put
 * the start and end address of heap into that external RAM.
 *
 * Note that memory MUST be contiguous between the start and
 * end point of the heap. If there are gaps in that memory
 * then malloc() will hard fault when it tries to use
 * memory from the gap.
 */
#include <stdint.h>
#include <errno.h>
#include <malloc.h>
#include "../util/util.h"

#define MAX_STACK_SIZE  8192

void local_heap_setup(uint8_t **start, uint8_t **end);

#pragma weak local_heap_setup = __local_ram

/* these are defined by the linker script */
extern uint8_t _ebss, _stack;

static uint8_t *_cur_brk = NULL;
static uint8_t *_heap_end = NULL;

/*
 * If not overridden, this puts the heap into the left
 * over ram between the BSS section and the stack while
 * preserving MAX_STACK_SIZE bytes for the stack itself.
 */
static void
__local_ram(uint8_t **start, uint8_t **end)
{
    *start = &_ebss;
    *end = (uint8_t *)(&_stack - MAX_STACK_SIZE);
}

/* prototype to make gcc happy */
void *_sbrk_r(struct _reent *, ptrdiff_t );

void *_sbrk_r(struct _reent *reent, ptrdiff_t diff)
{
    uint8_t *_old_brk;

    if (_heap_end == NULL) {
        local_heap_setup(&_cur_brk, &_heap_end);
    }

    _old_brk = _cur_brk;
    if (_cur_brk + diff > _heap_end) {
        reent->_errno = ENOMEM;
        return (void *)-1;
    }
    _cur_brk += diff;
    return _old_brk;
}
```

This is a function that could be built into the STM32 version of libopencm3, provide better malloc support by default and allow for the user to override where the heap goes by defining their own `local_heap_setup()` function. I did that in my DMA2D test code (which uses the SDRAM) like this:
```
/*
 * relocate the heap to the DRAM, 10MB at 0xC0000000
 */
void
local_heap_setup(uint8_t **start, uint8_t **end)
{
    console_puts("Local heap setup\n");
    *start = (uint8_t *)(0xc0000000);
    *end = (uint8_t *)(0xc0000000 + (10 * 1024 * 1024));
}
```
And I tested without defining the function, the heap appeared in the ram segment as expected, with the local_setup defined it appeared in the SDRAM memory as expected.

karlp, 2018-07-14:
Do you want to just move this to a wiki page?

ChuckM, 2018-07-14:
That probably makes the most sense.

ChuckM, 2018-07-14:
For your perusal https://github.com/libopencm3/libopencm3/wiki/Using-malloc-with-libopencm3

ChuckM, 2018-07-26:
With the code on the wiki I'm guessing we should close this.

closed 2018-07-26 by ChuckM