libopencm3/libopencm3 #950: USB corrupts data when using triple VCP

xtompok, 2018-08-02:
When using [pill_serial](https://github.com/satoshinm/pill_serial) or custom written code for triple usb to serial converter, the data sended to the last virtual com port (VCP) are corrupted. Instead of real data it comes something like `<0xd0><0x01><0xd0><0x01><0xd0><0x01>`. 

- changing  the order of VCP leads to change of the affected port, it is the last
- exchanging interrupt and bulk endpoints had no effect
- tested two BluePill boards and one custom board with the same results
- tested the pill_serial libopencm3 and current libopencm3 with same results

- tested GCC 4.8.4 (Jessie gcc-arm-none-eabi), GCC 4.9.3 (Ubuntu 16.04, gcc-arm-none-eabi) and GCC 7.3.0 (Ubuntu 18.04, gcc-arm-embedded)
- not tested gcc-arm-none-eabi on Ubuntu 18.04 because of broken USB support (compilable, but not working)

Has anyone similar issues? Can somebody try to test it with one of mentioned setup?

I tried to trace the problem by printing contents of the data buffer but at least just before entering ep_write_packet everything looks ok.

karlp, 2018-08-03:
Further, the output handling https://github.com/satoshinm/pill_serial/blob/master/src/usbuart.c#L230 looks _very_ dubious.  You're calling the blocking synchronous uart write, for _each byte_ inside the usb irq handler, (because you're trying to call poll() from your usb irq)

I don't imagine that will work very well.

xtompok, 2018-08-03:
I did some more investigation. I already tried increasing control buffer but it didn't help. I also tried to make minimal working code with 6 bulk endpoints (3 in and 3 out) but the 6th also did not work. 

Now I tried decreasing CDCACM_PACKET_SIZE to 32 and now everything works ok. Can somebody please explain, what is going inside and how to find it from outside?

karlp, 2018-08-20:
This was caused by #880.  using packet size of 32 would have avoided the bad buffer calculation.  You should be fine now tha tit has been merged.

closed 2018-08-20 by karlp