libopencm3/libopencm3 #954: stm32l011 linker script generator support

xor-gate, 2018-08-22:
Hi all,

I'm using a STM32L011 Nucleo-32 devkit with a STM32L011K4T6 (32pins) but it seems to not be supported in the ld/devices.data. I have added it myself and it works fine, as it seems there is no matching device when i specify `DEVICE = stm32l011` from the project Makefile.

This line I have added: `stm32l011* stm32l0 ROM=16K RAM=2K`. I'm not sure if I specify it incorrect or it is just plain not supported.

Kind regards,
Jerry Jacobs

karlp, 2018-08-22:
You're meant to do DEVICE=STM32L011K4T6 (any case), not just "stm32l011" 

Without the full name, you wouldn't be able to distinguish from the stm32l011k3t6, with only 8k flash.

closed 2018-08-22 by karlp

xor-gate, 2018-08-22:
That explains everything. Sorry for the noise.

karlp, 2018-08-22:
No problem, feel very welcome to report issues.