libopencm3/libopencm3 #955: STM32F4 - how to enable I/O compensation cell

dragonnn, 2018-08-23:
Hi,
I can't find any information in documentation or code how to enable I/O compensation cell. It is need if you want to use GPIO_OSPEED over 50MHz. I even can not find equivalent of SYSCFG_CMPCR_CMP_PD. I enabled RCC_SYSCFG but have no idea how to go further.
Hope someone can help me a little bit.

karlp, 2018-08-23:
Looks like at present only the register itself is defined.
```
SYSCFG_CMPCR |= 1; /* turn on */
while (!(SYSCGR_CMPCR & (1<<8))); /* wait for ready */
```

A PR to add those would be fine. I've left this open and retitlted it to track that these two bit definitions are missing.

karlp, 2018-08-23:
Also: this applies to F2, F4, F7.  For some reason the registger define is included for l1 and f3 as well, it should be pulled up to the separate parts.

dragonnn, 2018-08-23:
SYSCFG_CMPCR_CMP_PD should be  0x00000001 and SYSCFG_CMPCR_READY 0x00000100
I did find out that I need to enable not RCC_SYSCFG but SCC_SYSCFG, to get it working. What is the difference between RCC and SCC?

karlp, 2018-08-23:
Whether you need rcc or scc is up to you and your system.  SCC vs RCC is the "... in low power mode", so all the xxxLPENR registers in your reference manual.