libopencm3/libopencm3 #956: Consider initializing SCB_VTOR in reset_handler()

mattande, 2018-08-23:
User linker scripts may relocate the program ROM origin (for example when implementing separate bootloader and program code) which requires setting the vector table address in the reset handler.

The SCB_VTOR could be set as the first line of a user main() function, however this is too late if interrupts are enabled in c++ constructors.

Setting up the vector table prior to calling user's main() is also more consistent with other CM3 libraries users may be porting from.

Suggest a change similar to as follows. I can submit a pull request if this is something the project would incorporate.

```
--- a/lib/cm3/vector.c
+++ b/lib/cm3/vector.c
@@ -78,6 +78,9 @@ void __attribute__ ((weak)) reset_handler(void)
        /* Enabled by default on most Cortex-M parts, but not M3 r1 */
        SCB_CCR |= SCB_CCR_STKALIGN;

+       /* Set Vector Table address */
+       SCB_VTOR = (uint32_t) &vector_table;
+
        /* might be provided by platform specific vector.c */
        pre_main();
```

Initializing the SCB_VTOR is done in lib/vf6xx/vector_chipset.c where the default linker script places the vectors in RAM, which could then be removed.

zyp, 2018-08-27:
I would argue that when you introduce a bootloader between hardware reset and the application firmware, it's the bootloader's responsibility to provide a correct environment for the application firmware to start in, e.g. by setting SCB_VTOR before jumping to the application firmware's reset vector.

I would in general advise against doing too much in global constructors, you'll quickly find yourself in larger headaches than this due to the largely undefined order of execution. If you still need to set SCB_VTOR before the constructors run, I suggest doing that in another constructor and giving it an explicit priority to run before any other constructors. (Look up the `constructor` and `init_priority` attributes.)

mattande, 2018-08-30:
Agree that there are enough caveats here that a generic ```SCB_VTOR = (uint32_t) &vector_table;``` statement in the reset vector is probably not a good idea.

Not saying its a good practice but it is technically valid for interrupts to be enabled in global constructors (arduino-like firmware comes to mind).

Basically I am looking for a way to do pre_main() type of stuff on an application or target specific basis. Currently my only option is to modify the library and keep a local branch with those changes, which is not ideal.

What about adding a call to a ```user_pre_main()``` after the call to ```pre_main()``` with a default weak empty implementation? For example, porting from ST's HAL library provides a call ```HAL_MspInit()``` for this purpose.

zyp, 2018-08-30:
Like I mentioned in my previous comment, you can already do this through the existing constructor mechanism. Giving a function the `constructor` attribute will add it to the initializer list, which will make it get called along with all the C++ constructors. Assigning an explicit priority to it should place it first in the list, before all the constructors without a specified priority.

Alternatively, you can use an actual C++ constructor, and use the `init_priority` attribute on the instance to put it at the top of the list in the same manner.

mattande, 2018-08-30:
Thanks, using the ```constructor``` attribute should work for my application.

closed 2018-08-30 by mattande