libopencm3/libopencm3 #962: env.exe - Application Error

skxo, 2018-09-03:
Hi,

I try to compile on windows 10.
And I obtain the following window error env.exe
The application was unable to start correctly (0xc0000142)
Click OK to close the application

In the shell, I have this after make command:
 GENHDR  include/libopencm3/stm32/f0/irq.json
make: *** [include/libopencm3/stm32/f0/irq.json.genhdr] Error -1073741502

Any idea what it happens here on windows?
Regards

[image]

karlp, 2018-09-03:
what shell are you using?   If you've followed https://github.com/libopencm3/libopencm3#prerequisites for windows, I don't have anything new to add. I don't use windows personally and can't diagnose it any further.

karlp, 2018-10-20:
no feedback, no way forward

closed 2018-10-20 by karlp