libopencm3/libopencm3 #964: Create pwm signal, using alternative functions of the GPIO's

RabbeS, 2018-09-09:
Hi,

I've been trying to generate a pwm signal to a gpio pin for quite some time.
Therefor I would like to use the alternative functions, but don't know which timer settings I have to make. Unfortunately, I haven't  found any example.

Can you please explain me, what I have to adjust at the timer?

I'm using a **STM32F765VGT6**.
The pwm signal should be applied to the gpio pin **PE13**. According to the data-sheet I need the **AF1** of the pin. How do I have to setup **TIMER 1, CH3** to generate that signal?

The gpio pin is initialized as follows:
```
  rcc_periph_clock_enable(RCC_GPIOE);
  gpio_mode_setup(GPIOE, GPIO_MODE_AF, GPIO_PUPD_NONE, GPIO13);
  gpio_set_af(GPIOE, GPIO_AF1, GPIO13);
  gpio_set_output_options(GPIOE, GPIO_OTYPE_PP, GPIO_OSPEED_100MHZ, GPIO13);
```

Thanks for your help & best regards
Stephan

FabianInostroza, 2018-09-20:
I added a STM32F4DISCO pwm example, look here:
https://github.com/FabianInostroza/libopencm3-examples/tree/master/examples/stm32/f4/stm32f4-discovery/pwm

It should be not difficult to adapt to your MCU.

jacobrosenthal, 2018-09-20:
Sorry for the quick hijack. Anybody done soft spi/Twi?

RabbeS, 2018-09-21:
Thank you @FabianInostroza,

I just recognized, that timer 1 is an advanced-control timer. 
And in this case I need that method:
`timer_enable_break_main_output(TIM1);`

closed 2018-09-22 by RabbeS