libopencm3/libopencm3 #965: STM32F7 Makefile fails to recognize support for Cortex-M7 MCU

Derppening, 2018-09-10:
Whenever I am building the library, this message always pops up:
```
  BUILD   lib/stm32/f7
Makefile:71: Your toolchain doesn't support -mcpu=cortex-m7.
Please use gcc-arm-embedded 4.8 2014q3 or newer. Skipping this sub-library.
```
However, using `arm-none-eabi-gcc -mcpu=cortex-m7 -Q --help=target` shows that the target is indeed supported:
```
The following options are target specific:
...
  -mcpu=                                cortex-m7
...
```
I am somewhat certain that this is due to the way the check is conducted in `lib/stm32/f7/Makefile`:
```
ifneq ($(shell $(CC) --help=target | grep -q '\<cortex-m7\>'; echo $$?),0)
  $(warning $(MISSING_CPU))
```
When the command is ran with version `7-2018-q2-update`, a list of all known ARM CPUs is shown. However, with ArchLinux's version `8.2.0`, the list no longer appears. I am not sure whether this is a difference due to the version difference, or the embedded toolchain are modified as such.

If required, I can attach the full output of both compilers.

RabbeS, 2018-09-10:
Hi,

I've got the same problem.

Watch the line underneath of -mcpu.
It has to be:
`-- FPU         : hard-fpv5-sp-d16 `

karlp, 2018-09-10:
Now that it's 2018, we can maybe jsut drop the check.  Back when it was added there were plenty of systems where a toolchain supported m7 was hard to get.

closed 2018-09-10 by karlp