libopencm3/libopencm3 #968: stm32f1: USB enumeration delay/fail when EP0 size <= 16

FabianInostroza, 2018-09-20:
When the EP0 size (bMaxPacketSize0 in descriptor) is 16 or 8 in Linux there is a delay of about 7s before enumeration and enumeration fails on Windows 10, device manages says that the device descriptor is invalid.

I'm testing the cdcacm example in the stm32-h103 folder and a similar program in a stm32f105 device.

karlp, 2018-09-20:
please make or share the exact example code you are testing and failing.  A standalone github repo, anything, but something that can be recreated easily.  What is the behaviour on linux or osx? does it only fail on windows?

Are you running the latest examples repository? are you sure you have the recent usb fixes there?

FabianInostroza, 2018-09-20:
Hi karlp, see the following repo: https://github.com/FabianInostroza/libopencm3_usb_enum_problem
Let me know if you need more information.
The only modifications I did to the cdcacm example was the clock setup, the bMaxPacketSize0 value and the code to force reenumeration on reset.

I don't have OSX to test.

FabianInostroza, 2018-09-20:
I tested the demo in this library https://github.com/dmitrystu/libusb_stm32, it uses EP0 size = 8 and it doesn't have this issue.

karlp, 2018-09-24:
I confirm there's an issue.  On my linux host, with stm32l1, and ep0 of 16:
```
Sep 24 21:05:36 beros kernel: usb 1-4.2: new full-speed USB device number 13 using xhci_hcd
Sep 24 21:05:47 beros kernel: usb 1-4.2: device descriptor read/all, error -110
Sep 24 21:05:47 beros kernel: usb 1-4.2: new full-speed USB device number 14 using xhci_hcd
Sep 24 21:05:52 beros kernel: usb 1-4.2: Device not responding to setup address.
Sep 24 21:05:52 beros kernel: usb 1-4.2: Device not responding to setup address.
Sep 24 21:05:52 beros kernel: usb 1-4.2: device not accepting address 14, error -71
Sep 24 21:05:52 beros kernel: usb 1-4-port2: attempt power cycle
Sep 24 21:05:53 beros kernel: usb 1-4.2: new full-speed USB device number 15 using xhci_hcd
Sep 24 21:05:53 beros kernel: usb 1-4.2: New USB device found, idVendor=cafe, idProduct=cafe, bcdDevice= 0.01
Sep 24 21:05:53 beros kernel: usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 24 21:05:53 beros kernel: usb 1-4.2: Product: Gadget-Zero
Sep 24 21:05:53 beros kernel: usb 1-4.2: Manufacturer: libopencm3
Sep 24 21:05:53 beros kernel: usb 1-4.2: SerialNumber: stm32l1-generic
```

karlp, 2018-09-24:
I'd like to diagnose this further, I've asked a friend to see what's happening with a usb analyser.

In the meantime, "stop doing that" :)  Do you have any particular reason to use 16 or 8 bytes for EP0?  (Other than, "the spec allows it")

FabianInostroza, 2018-09-24:
I don't have any reason to use an EP size minor than 32. By chance I found the problem and was trying to fix it for fun and learning.
I have been analyzing the problem and found that after the first reset of the interface the host requests the device descriptor and after the device sends the first packet, the host starts sending out tokens (status stage) but the peripheral is sending NAK and the driver doesn't receive interrupt.
Then the get descriptor request times out after 5s and seems that for linux the first bytes of the descriptor is enough to assign an address and continues the enumeration, issuing other reset and requesting the descriptor again, windows seems to no tolerate this.

I analyzed the enumeration process  of a FT232RL, a FS device with bMaxPacketSize0 = 8 and it ACKs the status stage that the driver in libopencm3 is ignoring/not receiving.

FabianInostroza, 2018-09-24:
I pushed a sigrok/pulseview capture during enumeration to the sample repo, there are some errors due to sampling frequency and/or test leads but it shows the problem.

FabianInostroza, 2018-09-24:
From usb specs, 5.5.3 Control Transfer Packet Size Constraints

> In order to determine the maximum packet size for the Default Control Pipe, the USB System Software
> reads the device descriptor. The host will read the first eight bytes of the device descriptor. The device
> always responds with at least these initial bytes in a single packet. After the host reads the initial part of the
> device descriptor, it is guaranteed to have read this default pipe’s wMaxPacketSize field (byte 7 of the
> device descriptor).

karlp, 2018-09-24:
yeah, I know it's allowed, and it's definitely a bug, I was just curious if there was any _reason_ for it :)

zyp, 2018-09-25:
I've looked into this a bit and to me it seems like it's caused by deliberate non-specified behavior on the host, designed to deal with certain devices that also doesn't comply with the spec.

Here's a trace I did last night with MPS set to 16:
[image]

First capture is with macos. Here we see the host is first requesting the first 8 bytes, learns the MPS from it and then correctly reads the whole descriptor split into 16 byte chunks.

Second capture is with linux. Here we see linux sends a SETUP packet for a 64-byte read and device replies with the first 16-byte chunk. While device is waiting to send the second chunk, host believes the data stage is over and tries to send the status OUT packet, so neither gets accepted by the other side and the control request times out after five seconds. Even though the request partially timed out, linux appears to have learned the MPS from it and subsequently enumerates the device correctly.

Somebody in the IRC channel linked [this article](https://blogs.msdn.microsoft.com/usbcoreblog/2009/10/30/how-does-usb-stack-enumerate-a-device/#_First_Device_Descriptor), which suggests that Windows is doing more or less the same and offers some insight into why. I'll try to make a trace of enumeration on Windows tonight to see if I can figure out why it fails.

I suspect that this issue might have been exacerbated by the flow control improvements that got merged recently. Before the change, I guess that the status OUT packet would have been accepted even when it was unexpected.

I'm not sure what we should do about this issue. If it was only about the timeout, it would have been fair enough to say that you'll have to live with it when you insist on using a less common MPS, but since it makes enumeration fail completely on Windows, we should probably do something.

The cleanest solution from an outside perspective would be just accepting the status OUT stage after the first chunk, but this is something which we would only want to do for this particular request and that introduces a bunch of special casing in the control state machine.

The simplest workaround from a code perspective would be to always only send one chunk when the host requests 64 bytes, since hosts appears to request the exact 18 bytes when they want the whole thing. I.e. if the host requests 64 bytes and MPS is 16, we can just assume the host wanted 16 bytes. This breaks if the assumption is untrue though. To get around that, we could do this only for the first device descriptor request after each reset.

zyp, 2018-09-25:
I did some looking through Linux' `drivers/usb/core/hub.c` which is responsible for the host side of the initial enumeration. It looks consistent with the analysis in my previous post.

```
 * As of 2.6.10 we introduce a new USB device initialization scheme which
 * closely resembles the way Windows works.  Hopefully it will be compatible
 * with a wider range of devices than the old scheme.  However some previously
 * working devices may start giving rise to "device not accepting address"
 * errors; if that happens the user can try the old scheme by adjusting the
 * following module parameters.
```

```
 * NOTE:  If use_new_scheme() is true we will start by issuing
 * a 64-byte GET_DESCRIPTOR request.  This is what Windows does,
 * so it may help with some non-standards-compliant devices.
 * Otherwise we start with SET_ADDRESS and then try to read the
 * first 8 bytes of the device descriptor to get the ep0 maxpacket
 * value.
```

```c
buf->bMaxPacketSize0 = 0;
r = usb_control_msg(udev, usb_rcvaddr0pipe(),
	USB_REQ_GET_DESCRIPTOR, USB_DIR_IN,
	USB_DT_DEVICE << 8, 0,
	buf, GET_DESCRIPTOR_BUFSIZE,
	initial_descriptor_timeout);
switch (buf->bMaxPacketSize0) {
case 8: case 16: case 32: case 64: case 255:
	if (buf->bDescriptorType ==
			USB_DT_DEVICE) {
		r = 0;
		break;
	}
	/* FALL THROUGH */
default:
	if (r == 0)
		r = -EPROTO;
	break;
}
```
Here we see that receiving a valid value for bMaxPacketSize0 would be treated as a success even if the request timed out.

zyp, 2018-09-26:
I made another trace, this time plugging it into a Windows computer:
[image]

Unlike Linux, it seems to not accept the partial data that was received before the timeout and instead does the three retries the article I linked mentioned, before giving up.

We also seem to have a bug here on the device side, returning only four bytes to requests after the first. I don't think this matters for the actual problem, the main symptom is still status OUT timeout. This bug might be dwc_otg specific, I've only tested with a F407.

zyp, 2018-09-26:
I tried the simple workaround I mentioned earlier:

```diff
diff --git a/lib/usb/usb_standard.c b/lib/usb/usb_standard.c
index 4ce7a8b1..d71db059 100644
--- a/lib/usb/usb_standard.c
+++ b/lib/usb/usb_standard.c
@@ -161,6 +161,9 @@ usb_standard_get_descriptor(usbd_device *usbd_dev,
 
        switch (usb_descriptor_type(req->wValue)) {
        case USB_DT_DEVICE:
+               if(*len == 64) {
+                       usbd_dev->control_state.req.wLength = *len = usbd_dev->desc->bMaxPacketSize0;
+               }
                *buf = (uint8_t *) usbd_dev->desc;
                *len = MIN(*len, usbd_dev->desc->bLength);
                return USBD_REQ_HANDLED;
```

Now it enumerates fine with both Linux and Windows respectively:
[image]

FabianInostroza, 2018-09-26:
@zyp, the workaround you mention works for enumeration buy may break user software that reads descriptors requesting 64 bytes. I don't know of any software that does this, but it is possible. See the following python script:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import usb

dev = usb.core.find(idVendor=0x0483, idProduct=0x5740)

print dev.ctrl_transfer(bmRequestType=0x80, bRequest=6, wValue=0x0100, wIndex=0, data_or_wLength=64)
```

I think a better option would be to just ACK the stage out the host is sending (at least when there is no address set). Some code I haven't found is disabling the reception of stage out packets, so the firmware never knows the host switched to the stage out and the peripheral is just NACKing those packets.

zyp, 2018-09-26:
> Some code I haven't found is disabling the reception of stage out packets, so the firmware never knows the host switched to the stage out and the peripheral is just NACKing those packets.

Yeah, I wrote that code. See #785. :)

As I wrote in my first comment on this issue:

> The cleanest solution from an outside perspective would be just accepting the status OUT stage after the first chunk, but this is something which we would only want to do for this particular request and that introduces a bunch of special casing in the control state machine.

I'm aware my three-line workaround is hackish, feel free to have a go at making a more elaborate and better one. :)

FabianInostroza, 2018-10-08:
@zyp, adding `usbd_dev->current_address == 0 ` to the `if` limits the scope of the workaround to just the enumeration, that seems good enough for me,  what do you / other people think?

zyp, 2018-10-08:
That sounds reasonable to me.

karlp, 2019-06-03:
Seems reasonable enough.  If you ever develop that as a PR please mention this ticket by URL in the comment, but until then, I feel this ticket has enough information in it for other readers, and there's no further work being on it.

closed 2019-06-03 by karlp