libopencm3/libopencm3 #971: Miniblink on NUCLEO-L432KC

lixnig, 2018-09-24:
So I'm pretty new to the STM32 MCU, and I've been going through the Beginning STM32 book by Warren Gay.  I have about a dozen of the NUCLEO-L432KC boards around that I use for making LCD demos with the MBED interface, so I figured i'd trying using one of these for the book tutorials since they use the same MCU, but when modifying the miniblink.c file to use GPIO3 on Port B (what I believe to be the user controllable LED on these boards), I get no output on the LED when flashed.  Even when simply turning LED on, I seem to get no output.  

Is anyone familiar with using this board and libopencm3?  Any ideas of where I might be going astray?  Source and Makefiles below.

[Makefile.txt](https://github.com/libopencm3/libopencm3/files/2412547/Makefile.txt)
[Makefile_incl.txt](https://github.com/libopencm3/libopencm3/files/2412548/Makefile_incl.txt)
[miniblink.txt](https://github.com/libopencm3/libopencm3/files/2412549/miniblink.txt)

karlp, 2018-09-24:
Try using miniblinks from https://github.com/libopencm3/libopencm3-miniblink/blob/master/boards.stm32.mk#L132  The L432 nucleo is there.

karlp, 2018-09-24:
FOr starters, your makefiles seem to be targeting stm32f1, instead of l4.

lixnig, 2018-09-24:
Thanks for the suggestion, I'll check out the miniblinks.  I think maybe I am a bit confused here.  On one side of the PCB is an STM32F103CBT6 and on the other is an IC labeled L432C6U.  I had assumed I was compiling for the F1 part based on the board schematic.  Are you saying this should actually be an L4?

[image]
[image]

karlp, 2018-09-24:
Please the documentation for NUCLEO32 device :)  The stm32f1 implements the STLink onboard debugger.  The L4 is your target.

karlp, 2018-09-24:
the _name_ of the nucleo is the part it targets, they _all_ have the stm32f1 on them to implement the debugger.

closed 2018-09-24 by karlp