libopencm3/libopencm3 #982: [stm32/can] Get number of pending messages

jsphuebner, 2018-10-18:
I added a function for getting the number of pending messages in the given fifo. Seems to be the only way to figure out whether it's worthwhile calling can_receive().
Any chance to have it added to the library?

```
/** @brief CAN Transmit Message

@param[in] canport Unsigned int32. CAN block register base @ref can_reg_base.
@param[in] fifo Unsigned int8. FIFO id.
@returns int 1, 2 or 3 if messages are pending in given fifo, 0 otherwise.
 */
uint32_t can_fifo_pending(uint32_t canport, uint8_t fifo)
{
	if (fifo == 0) {
		return CAN_RF0R(canport) & CAN_RF0R_FMP0_MASK;
	} else {
		return CAN_RF1R(canport) & CAN_RF1R_FMP1_MASK;
	}
}
```

karlp, 2018-10-20:
Sure, seems reasonable.  Is this the api you want? Should it be built into can_receive?  should can_receive be releasing mailboxes?  Just submit it as a PR so I can merge it easily.

jsphuebner, 2018-10-20:
It could be integrated into can_receive such that it returns a bool. So true when it actually received a message, false otherwise. Not sure what can_receive() currently does when the FIFO is empty.

No, the embarrassing part: I have no idea how pull requests work. Any pointers? The github documentation was merely confusing.

karlp, 2018-10-20:
Well, presumably you've been using this code?  You're in a much better position to know what can_receive() does when empty than I am :)

jsphuebner, 2018-10-21:
Yes just tried that. It returns some random data:
id: 12b79caa, len: 6, data[0]: 7a0ef27c, data[1]: 554cf144
and other data on fifo1
So I think I will integrate the fifo count checking into can_receive()
Just how do I do that pull request? Do I even have write access?

karlp, 2018-10-21:
you fork the repository to your own github account, make changes and push to that repo, and then open a PR for it here.  See https://help.github.com/articles/creating-a-pull-request/ and the first two things "from a fork... and "abotu forks"