libopencm3/libopencm3 #983: USB fails to enumerate on EFM32HG

dlharmon, 2018-10-18:
reverting commit 26ab78a710b4cb7abcace769768c4781332b17ba restores enumeration. I also tested on STM32F4 and that worked fine with or without that commit in place.

Host OS is Debian Stretch.

I'm using custom hardware very similar to https://tomu.im The Tomu USB HID example works correctly on my hardware with the commit reverted and fails with the commit in place, so that is a good test. The repo is here:  https://github.com/im-tomu/tomu-quickstart You do need to replace their libopencm3 with this repo's head to get it to fail.

dmesg output when device attached:
```
[28824.863211] usb 1-2.2-port3: attempt power cycle
[28825.466603] usb 1-2.2.3: new full-speed USB device number 119 using xhci_hcd
[28830.562753] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[28835.938587] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[28836.146420] usb 1-2.2.3: device not accepting address 119, error -62
```

karlp, 2018-10-18:
I have a happy gecko board I'll retest with.  There's also hg support in gadget0 which is what I> test with.

karlp, 2018-10-18:
yeah, fails here too. lame shit.

zyp, 2018-12-26:
I just debugged an issue with my own stack on stm32f446 which might be related to this issue.

I see that on f446 the hardware doesn't always trigger the SETUP complete event when we expect it to, causing enumeration to get stuck and time out. I suspect this is related to differences between revisions of the dwc_otg core and would therefore also occur on other chips with the same revision.

It might be useful to start making a list of which revision each chip with the dwc_otg core contains, along with what behavior they show. I'm afraid we'll have to start adding alternate code paths for different revisions, but I hope we can find one solution that works for all.