libopencm3/libopencm3 #1000: Merge project with yours

alejandro-colomar, 2018-12-19:
Hi, I started some months ago in embedded systems in a university course.

I didn't like HAL libraries too much, so I looked for a free library and found this one.  However, it isn't maintained too much for the STM32L4, so the professor recommended me to use the HALs and I did so.

Now I have a more or less working library for some things on that board. (On January, I have to finish the project for that course).

I would like you to see if my library is of any help for this project, and if so, contribute to it with the code. :)

karlp, 2018-12-19:
by all means, just submit them as chunks, and follow the existing styles, just open PRs as you see fit.

Your code all appears to setill be using st's HAL though, so it won't really merge or be applicable at this point.  I hope this has answered your question though.

just FYI, it _is_ maintained for L4, there just hasn't been a lot of push to unify l4 peripherals properly where they are the same as on other boards.

karlp, 2019-06-03:
Closing as there doesn't seem to be much need for a special ticket for this.  As mentioned, feel welcome to submit PRs for portions you have developed.

closed 2019-06-03 by karlp